iNavFlight/inav #3495: V2.0.0 RC2 Flight Time alsways 00:00

Gregmaan, 2018-07-01:
In the OSD Display the flight time shows always 00:00 during flight and in summary, using the "On Time / Fly Time" Timer in OSD setup.
diff:
https://pastebin.com/jF22efqs

shellixyz, 2018-07-02:
I'm using the "On Time / Fly Time" timer and can't reproduce the issue.

Not directly related to this issue: your battery capacity warning/critical thresholds are not configured correctly. Warning should be greater than critical. They represent the remaining mAh/mWh you want the thresholds to be triggered at. Please read the [battery management documentation](https://github.com/iNavFlight/inav/blob/development/docs/Battery.md#battery-capacity-monitoring) for more information.

renatobfa, 2018-07-03:
> In the OSD Display the flight time shows always 00:00 during flight

Mine too.. Im flying with an Omnibus F4 Pro, I've also had this problem, showing always 00:00 after arming and in flight.

teckel12, 2018-07-03:
Also running Omnibus F4 Pro and my "On Time / Fly Time" OSD is just fine.

Gregmaan, 2018-07-03:
I think I found something:
If I start the flight with launch mode, the timer works.
So there could be an issue with flight detection without using launch mode.

teckel12, 2018-07-03:
@Gregmaan I launch with switch.

Gregmaan, 2018-07-03:
@teckel12 Do you mean launch mode on switch?
Before I did not use any launch mode ( with the issue).
Now I use Launch mode as feature and fly time is fine.

shellixyz, 2018-07-03:
Thank you for the report it will be fixed in the next RC. Indeed it happens when not using launch mode on fixed wing.

fiam, 2018-07-07:
Fixed by @shellixyz on #3509, will be on RC3

PR #3509 (2018-07-07): Fix flight timer for fixed wing when not using launch mode
Fixes #3495
closed 2018-07-07 by fiam