iNavFlight/inav #3497: Fport rssi scale

VadimGromov, 2018-07-01:
# version
# INAV/OMNIBUS 2.0.0 Jun 29 2018 / 12:33:58 (a8415e89c)

R9mini connected via fport. The rssi max value is 50% and does not changes with rssi_scale.

shellixyz, 2018-07-03:
Please give us more details:
- Config diff/dump
- How is the RSSI sent ? SBus channel ? Which one ?

VadimGromov, 2018-07-03:
RSSI sent via fport. ADC and channel are disabled.
# diff 

# version
# INAV/OMNIBUS 2.0.0 Jun 29 2018 / 12:33:58 (a8415e89c)

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 3 1000 2000 1500 -100 

# feature
feature -TX_PROF_SEL
feature -BLACKBOX
feature MOTOR_STOP
feature GPS
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH

# beeper

# map
map TAER

# serial
serial 0 2 115200 38400 0 115200
serial 1 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 2 0 1300 2100
aux 1 8 1 1675 2100
aux 2 19 1 1300 1700
aux 3 35 0 1700 2100
aux 4 27 2 1300 1700
aux 5 28 2 1700 2100

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 22 12 V
osd_layout 0 1 1 10 V
osd_layout 0 4 8 6 H
osd_layout 0 11 1 11 V
osd_layout 0 12 1 12 V
osd_layout 0 13 1 7 V
osd_layout 0 14 24 3 V
osd_layout 0 15 24 6 V
osd_layout 0 16 6 10 V
osd_layout 0 17 6 11 V
osd_layout 0 20 18 0 V
osd_layout 0 21 0 0 V
osd_layout 0 22 14 4 V
osd_layout 0 23 12 3 V
osd_layout 0 25 23 5 V
osd_layout 0 26 24 7 V
osd_layout 0 31 24 2 V
osd_layout 0 33 6 2 V
osd_layout 0 35 1 9 V
osd_layout 0 40 12 2 V
osd_layout 0 51 12 1 V
osd_layout 0 52 12 0 V

# master
set looptime = 2000
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 40
set acc_hardware = MPU6000
set acczero_x = 52
set acczero_y = -26
set acczero_z = -407
set accgain_x = 4087
set accgain_y = 4053
set accgain_z = 4039
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set platform_type = AIRPLANE
set model_preview_type = 8
set small_angle = 180
set fixed_wing_auto_arm = ON
set disarm_kill_switch = OFF
set gps_provider = UBLOX7
set gps_sbas_mode = AUTO

# profile
profile 1

set fw_p_pitch = 6
set fw_i_pitch = 9
set fw_ff_pitch = 52
set fw_p_roll = 6
set fw_i_roll = 6
set fw_ff_roll = 49
set tpa_rate = 20
set tpa_breakpoint = 1600
set roll_rate = 28
set pitch_rate = 14

# battery_profile
battery_profile 1

Droneracer101, 2018-07-09:
I am having the same issue with iNAV 1.9.1.  F.Port natively transmits RSSI, not analog, not on a channel. I have confirmed it works, but rssi_scale isn't working with it.  It shows 50% of the true value.  It does the exact same thing in Betaflight 3.4 but there I can use set rssi_scale =200 and then it scales perfectly with the RSSI displayed on the radio.  Doing the same thing in iNav doesn't seem to affect the displayed value with this input.  

This is what set rssi? shows me:
# set rssi
rssi_adc_channel = 0
Allowed range: 0 - 4

rssi_channel = 0
Allowed range: 0 - 18

rssi_scale = 200
Allowed range: 1 - 255

rssi_invert = OFF
Allowed values: OFF, ON

osd_rssi_alarm = 60
Allowed range: 0 - 100

osd_rssi_pos = 2071
Allowed range: 0 - 3071

For reference is here the same value I am using in Betaflight with F.Port.  The only difference is the rssi_scale seems to work.

# set rssi
rssi_channel = 0
Allowed range: 0 - 18

rssi_src_frame_errors = OFF
Allowed values: OFF, ON

rssi_scale = 200
Allowed range: 1 - 255
Default value: 100

rssi_offset = 0
Allowed range: -100 - 100

rssi_invert = OFF
Allowed values: OFF, ON

osd_rssi_alarm = 20
Allowed range: 0 - 100

osd_rssi_pos = 2284
Allowed range: 0 - 3071
Default value: 234

osd_stat_min_rssi = ON
Allowed values: OFF, ON

Also, rssi_scale does seem to work in iNAV if I have another rssi method selected. Just not with F.Port.  But now that I have it I really like FPort RSSI.  It seems to most accurately match the value on my radio.

fiam, 2018-07-19:
Fixed in #3613

closed 2018-07-19 by fiam

MrVinther, 2018-08-23:
Can confirm that this issue is still present in 2.0.0 final
Using Matek f722 board Frsky R9 mini using latest F-port non-EU firmware.

teckel12, 2018-08-24:
@MrVinther Are you setting the new `rssi_min` and `rssi_max` values?

MrVinther, 2018-08-24:
This results in 50% rssi in the configurator. I have 100% rssi on my Horus.
The old rssi_scale could be set to 200 which solved the problem.

# get rssi
rssi_adc_channel = 3
Allowed range: 0 - 4

rssi_channel = 16
Allowed range: 0 - 18

rssi_min = 0
Allowed range: 0 - 100

rssi_max = 100
Allowed range: 0 - 100

osd_rssi_alarm = 20
Allowed range: 0 - 100

MrVinther, 2018-08-24:
Im sorry for this. I understand now how to set the rssi_max value. Setting it to 50% as the max showed value in the configurator gives me 100%.

teckel12, 2018-08-24:
@MrVinther Yeah, no need for scale.  You simply set the min and max values to what they are (even if the max value is negative) and the code will take care of the rest.

jasc76, 2018-11-11:
Just as info:
Using:
Matek F405 Wing
Inav 2.0
R9 MM F.Port (Firmware FW-R9MINI-R9MM-F.PORT-180629)

I had to set rssi_channel = 0 and rssi_adc = 0 otherwise fport rssi would not show correctly. rssi_adc was set I think to 2 in the defaults.
thought I should share that

throttlenerd, 2019-11-18:
Same here, guys! Matek F722SE, R9MM. Flashed EmuFlight just to check the rssi_scale method -- works perfectly. Flashed iNav 2.3.0 RC2 (configurator is updated too) -- no, only rssi_min and rssi_max which is already maxed out at 100. But my R9MM is flashed with FLEX firmware and it outputs RSSI on channel 16, so I will just use it as an rssi_source.

agrestic1, 2020-05-11:
Same here! Matek F722SE, R9MM with FLEX firmware on iNAV 2.5.0 RC1
But using Channel 16 works fine!

skorokithakis, 2020-05-29:
For anyone looking at this thread for a solution, it looks like to solve this you can EITHER do:

```
set rssi_max = 50
```

OR, if your receiver sends RSSI back on channel 16:

```
set rssi_channel = 16
```

commit c34a408f (2018-07-16): Add rssi_offset, apply offset and scale to all protocols

This allows any possible received RSSI to be mapped to [0, 100],
no matter the range and offset of value emitted by the RX.

Also, apply the RSSI scale and offset to all RSSI sources except
via MSP (via MSP we expect the sender to be well behaved and send
only values in [0, RSSI_MAX_VALUE]).

Fixes #2898
Fixes #3497
Fixes #3607