iNavFlight/inav #3498: Possible fault when using EXBUS, lost link and no telemetry

bmadeira, 2018-07-01:
Today I finally tested my Inav setup, I using a airbot mini FC and a Jeti Radio using EXBUS on uart 3.

My problem, I lost radio link at about 40-60m but not due a link problem, because the rssi is always between  87-99% and the rssi on the radio is the same.

My question is could this be an issue with the protocol EXBUS from Jeti in Inav? I use Betalfight with EXBUS and similar receivers and it works great.

Another thing that is not working is telemery via EXBUS, again on Betaflight it's working

[image]

[image]

[image]

thomasekdahlN, 2018-07-30:
I have the same problem.

I have wired up a Matek F405 WING on UART2 (Signal wire to RX, and 2k4 resistor between RX and TX) to a Jeti RX Pin with EX-BUS enabled (An Assist REX 6 - see the irony - I only had a RX with gyro builtin available for testing - if I get this working I will not need Assist receivers;-)

iNav is receiving all channels perfectly with JETIEXBUS mode, but it does not transmit telemetry back to the transmitter. (Well, on the bench it seems to work at least)

I have tried both UART2 and UART4
I have tried with 2k4 resistor and without (it works the same without, NO telemetry, but as I understand the resistor should make telemetry work)
Tried changing Jeti RX Pin to PPM output to see if everything really was PPM, but then it stopped working in iNav. So I assume EX-BUS is sendt correctly from Jeti RX to iNav when enabling EX-BUS on the correct pin.

bmadeira, 2018-07-30:
For me it solved not to use EXBUS and only use PPM, in some receivers the PPM is in a diference pin from EXBUS. (also um must to use PPM Positive)
In my case the EXBUS works OK on the bench but flying it cashed 2 times because of radio signal lost, and that was at about 30-50 meter and in the radio it was almost at max rssi.

With PPM I haven't no problem.

thomasekdahlN, 2018-07-31:
With PPM you have given up on Telemetry (and gone analog)? Or do you get Jeti Telemetry to work also?  I think I will wait connecting up my Jeti gear to the FC's with iNav and use FrSky (Taranis) and FPort until the JETIEXBUS support is more mature. I was hoping EXBUS support was good and made Jeti an equal player i the drone world - but it is apparently not (yet) since most use.

closed 2018-10-13 by stale[bot]

RobinatiNav, 2023-03-25:
> For me it solved not to use EXBUS and only use PPM, in some receivers the PPM is in a diference pin from EXBUS. (also um must to use PPM Positive) In my case the EXBUS works OK on the bench but [...]

HI, I know this is old topic but let me ask you. Do you still fly Jeti and iNav together? If so, what protocol do you use? Thanks

thomasekdahlN, 2023-03-26:
Yes, we are many people flying Jeti with inav using exbus without any other problem than if you have more than 16 channels sendt via EXBUS inav will crash. If you only transmit 16 channels it will work fine.

RobinatiNav, 2023-03-26:
Thanks but we have different experience. I have DC-16 and Rx is setup for 16 channels, so I believe My setup is working inside the 16 channel range. We found out the last working EXBus was in iNav compiled with GCC9.3.1. Later versions of compiler somehow broke the protocol.