iNavFlight/inav #3499: Request for remote gain adjustments via aux channel

Hyprmtr, 2018-07-02:
There’s been a few issues with users complaining about very unstable flight conditions with FW stability modes especially in windy conditions. Of course, many of us already know we need to tune accordingly to our environment and not assume a preset is going to make magic happen in all weather conditions. An idea I have that other autopilots use may be a welcoming addition. 

A simple aux channel on a pot to P, I, or D(FF) roll, pitch, or yaw(selectable) would allow a much more efficient tune without having to land, adjust, test, land, adjust, test,...

If this has been submitted in the past or, has been requested and not enough interest to pursue, please pardon. 

Thank you and good day.

Arvycka, 2018-07-02:
You can already do that:
https://github.com/iNavFlight/inav/blob/master/docs/Inflight%20Adjustments.md
https://www.youtube.com/watch?v=NtkQ-bf9Ke8&t=347s

shellixyz, 2018-07-02:
As previously stated it is already possible and a lot of new types of adjustments have been added in 2.0. You can also now see all the adjusted values in the OSD and see which one is active (blinking). And also display the PID controllers outputs.

taileron, 2018-07-02:
still trying for a tailsitter having 4 times lower PIDs when flying fast forward and being able to switch back inflight for landing - needs a direct control of the PIDs from an aux channel like a tpa factor to achieve a 20% factor or better a linear control of PIDs in percentage with a tx slider or tx mixer.
Currently the only bad quality workaround is using rate profiles in combination with different tpa rates.

closed 2018-07-03 by DzikuVx