iNavFlight/inav #3506: V2.0.0 Aircraft diving in Cruise mode

renatobfa, 2018-07-03:
Its a Omnibus F4 Pro with INAV 2.0.0.

I've succesfuly flown my aircraft in V1.9.1 and in V2.0.0 with Angle + Altitude Mode + Heading Hold, and it worked fine in both versions....

But, in 2.0.0, when I tried to fly in CRUISE mode only (all other flight modes off), the plane kept diving like -2.0 m/s the whole time.

Anyone else had this problem?

teckel12, 2018-07-03:
I saw this when I upgraded without recalibration. Once I recalibrate, everything was stable. I suggest you recalibrate.

renatobfa, 2018-07-03:
The first thing i did when I installed V2.0.0 was to calibrate it... So, i dont think thats the problem. Besides, the Angle + Altitude Mode + Heading Hold works fine in V2.0.0, so perhaps the problem is in Cruise mode or some other thing.

teckel12, 2018-07-03:
@renatobfa For me, before I calibrated, my AccZ was always showing it dropping at like 50 feet per second, even when on my bench.  I'd verify that AccZ is zero on the bench as it really sounds a lot like the issue I had.

closed 2018-09-15 by stale[bot]