iNavFlight/inav #3508: V2.0.0 Scale "XX m" of Radar mode sometimes go crazy

renatobfa, 2018-07-03:
It's a Omnibus F4 Pro with INAV 2.0.0.

Radar mode, with "aircraft" in the middle and "home" moving, need some ajusting because the scale keeps changing from "320m" to "2.56 km", or "80 m" to "1.28 km" and the "home" icon apears just under the aircraft in those moments... it auto correcs after a few seconds
[image]

fiam, 2018-07-07:
@renatobfa could you send me privately a video between 30s after and before the problem? Being able to see how the scale was changing before reaching that value would help to determine the best way to fix the bug.

renatobfa, 2018-07-10:
@fiam , Hello friend! 

As you asked, i've sent to your email a clip soon before the Radar Mode Bugs occur... they happen 2 times in this clip, but the second time is very brief because I changed the OSD soon after the second time, but it is noticeble.

My guess is that the RADAR Scale goes crasy right when "home" is 90º to the LEFT or RIGHT in relation of the craft..... see for youself.

Keep up the good work. Thanks!

fiam, 2018-07-10:
@renatobfa Thanks for the video. I've looked at it and I'll fix it as soon as possible.

fiam, 2018-07-11:
Fixed in #3562

PR #3562 (2018-07-11): Check for the correct display coordinate when drawing POI
When checking the display, we were testing for (poiY, poiY) rather
than (poiX, poiY), which was specially visible when drawing across
the horizontal line over the map center, since the test was always
for (0, 0), causing the scale to go to the maximum value since
(0, 0) always contains the map center symbol.

Fixes #3508
commit 867e46eb (2018-07-11): Check for the correct display coordinate when drawing POI

When checking the display, we were testing for (poiY, poiY) rather
than (poiX, poiY), which was specially visible when drawing across
the horizontal line over the map center, since the test was always
for (0, 0), causing the scale to go to the maximum value since
(0, 0) always contains the map center symbol.

Fixes #3508
closed 2018-07-11 by fiam