iNavFlight/inav #3511: New feature request - Accelerometer on OSD

Artyum, 2018-07-03:
iNav 2.0 introduced lots of new OSD elements like RC Adjustable values or PID outputs. Could you consider adding to OSD some more variables corresponding to accelerometer reading. One of the reason to do that would be to show how the model is stress, proportional to speed during diving or acrobatics.

Artyum, 2018-07-08:
In fact i implemented the feature few days ago on devel branch. All i did, I put just z-axis on osd just to feel sense of it. Actually it was quite suprise to see 3.0g downforce on a loop. The ~2.3kg model weights about 7kg so wings and that plastic screws must withstand. I found it useful and fun feature for acro flying. I leave the developers the judgement if it is worth to add to inav.

shellixyz, 2019-05-24:
This feature has been implemented and is going to be available starting from 2.2-RC1, closing

PR #4696 (2019-05-12): g-force OSD item and stats
Take over from #3566. Closes #3511. Closes #4135 

Matching configurator update: https://github.com/iNavFlight/inav-configurator/pull/765
closed 2019-05-24 by shellixyz