iNavFlight/inav #3516: 2.0.0RC2 blackbox log missing gps time and timezone

davidngrc, 2018-07-04:
I am in iNav 2.0.0RC2 + MATEK F405 Wing
in the blackbox log file (in the sd card) (view by INAV Blackbox Explorer 3.0.2 )

please add the GPS date time to the black box log as well.
it currently only have the time from power on.

please also record the timezone as well which set by CLI tz_offset

ninja-zx11, 2018-07-04:
Yes that would be great if its possible.It will make it easy to locate or find.

stronnag, 2018-07-04:
The date is already recorded as UTC.
```
H Log start datetime:2018-07-04T14:30:39.058+00:00
```
```
H Log start datetime:2018-07-04T15:18:19.739+00:00
```
Logs copied to the file system from SD show the correct creation time:
```
$ ls -l /t/inav/2018-07-04_tri/
total 39264
-rw-r--r-- 1 jrh jrh 7387931 Jul  4 15:30 LOG00001.TXT
-rw-r--r-- 1 jrh jrh 7412021 Jul  4 15:47 LOG00002.TXT
-rw-r--r-- 1 jrh jrh 7008061 Jul  4 16:03 LOG00003.TXT
-rw-r--r-- 1 jrh jrh 6709931 Jul  4 16:18 LOG00004.TXT
```
(The file system shows current (daylight saving) time, it is consistent with the log.

ninja-zx11, 2018-07-04:
Is date and time corresponds to the time of creation by the Flight controller or when the logs are copied from the SD card or internal flash to the computer?
I meant that it will be nice to have date and time (when it was created by the FC) as a file name of the log.

stronnag, 2018-07-04:
As stated above, it the the log creation (start time) in UTC.  It is the responsibility of the host file system to maintain the time at which it was copied, in this case about 1.5 hours later:
```
$ stat /t/inav/2018-07-04_tri/LOG00001.TXT 
  File: /t/inav/2018-07-04_tri/LOG00001.TXT
  Size: 7387931   	Blocks: 14432      IO Block: 4096   regular file
Device: 821h/2081d	Inode: 22293571    Links: 1
Access: (0644/-rw-r--r--)  Uid: ( 1000/     jrh)   Gid: ( 1000/     jrh)
Access: 1980-01-01 00:00:00.000000000 +0000
Modify: 2018-07-04 15:30:38.000000000 +0100
Change: 2018-07-04 16:59:12.195594771 +0100
```
You could easily generate a small script to rename the files if you really wanted.
```
#!/bin/bash

for F in LOG*.TXT
do
 TIMET=$(stat -c %Y $F)
 T=$(date -d @$TIMET +%FT%T)
 mv $F BBL_$T.TXT
done
```
Voila!
```
$ ls -l 
total 39264
-rw-r--r-- 1 jrh jrh 7387931 Jul  4 15:30 BBL_2018-07-04T15:30:38.TXT
-rw-r--r-- 1 jrh jrh 7412021 Jul  4 15:47 BBL_2018-07-04T15:47:16.TXT
-rw-r--r-- 1 jrh jrh 7008061 Jul  4 16:03 BBL_2018-07-04T16:03:08.TXT
-rw-r--r-- 1 jrh jrh 6709931 Jul  4 16:18 BBL_2018-07-04T16:18:18.TXT
```

ninja-zx11, 2018-07-04:
Sorry for the confusion.Actually i meant like this in the attached picture.
For example in a picture,i have a one log file which i copied from internal flash to my pc and it contains multiple entries.It would be great if those different entries reflect the time and date of a flight.Thanks .
[image]

stronnag, 2018-07-04:
The time of the flight is recorded in the combined log for each indiviual entry already (indiviual H frames as above). This is a presentation issue and should be on a blackbox viewer issue, not an FC issue (because there is no FC issue here). The FC already stores all the data.

ninja-zx11, 2018-07-04:
Thanks stronnag.Yes you are right.This feature request should be on a blackbox viewer.Not sure what davidngrc meant.
BTW i am loving your mwptools!!

stronnag, 2018-07-04:
You might like the `mwptools/bbox-replay/bb_split.rb` script, which will split out a combined flash log into individual log files. And maybe I should add a clock date the mwp BB log chooser ... huum.

ninja-zx11, 2018-07-04:
Yes that would be great to have log entries showing clock and date along with duration in a mwp BB log chooser window !!.Thanks stronnag.

davidngrc, 2018-07-05:
I meant exactly blackbox log missing gps time and timezone
the date time @stronnag  mention here is a file attribute, it might changed by by someone.

for example
if I open this log file in a hex editor, and click save, the modify date will change, some virus will change this as well.
 
if I copy the file to other place, the new file create date will change.

we can store only one gps date time and time zone to the blackbox log, no need to store for every data row.

it is loke db design, u have a data table which store the original data, then u have another table call others, store like
attribute, vlaue
gps_date_time, 2018-07-01 15:20
time_zome, +08:00
aircraft_name, HD_Wing
Flight_Controller, F405 Wing

DzikuVx, 2018-07-05:
Like @stronnag stated, time is logged inside of the log
`H Log start datetime:2018-06-23T06:35:22.301+00:00`
It has nothing to do with file creation or modification date. 

To display this information in a blackbox viewer is a separate thing and has nothing to do with firmware

closed 2018-07-05 by DzikuVx