iNavFlight/inav #3518: Pre-Arm Check...rex X on Navigation is Safe

claytonbakerjr, 2018-07-05:
Hi Sorry
I seen an issue similarly titled but had to do with a GPS. In my case no matter what channel I select once I enable RTH the Navigation is Safe turns red. Once I delete it it goes back to green. In this state I cannot arm.
Im running Configurator 1.9.3 with a Furious FPV F35 flashed with iNav 1.9.1
Im running a TBS Crossfire TX module with a TBS Diversity RX. Both flashed with the latest versions.
In short in the Modes tab I am setting up my modes I lastly setup RTH. I armed and tried to rotate motor but no go, then looking in the Setup tab is when I seen the red X.
I can repeat this simply by enabling RTH in the modes tab.
The only modes I currently have setup is ARM, Manual and Horizon

claytonbakerjr, 2018-07-05:
Ok sorry I feel I found the solution.
set nav_extra_arming_safety = OFF
Just to clarify however this settings simply allows overriding safety checks
but in the same breath RTH will still work as intended?
Thanks

DzikuVx, 2018-07-05:
You can not ARM when RTH is enabled. How would that even supposed to work? `nav_extra_arming_safety` is not advised to be disabled

closed 2018-07-05 by DzikuVx