iNavFlight/inav #3523: uncontrolled Quadrocopter with 2.0

Dronitoni, 2018-07-05:
MATEK F222-STD from BG
INAV 2.0

My new Quadrocopter sometimes  shows a strange behavior.
He is wobbling around like a leave in the wind, but with full throttle.

This behavior is reproduceable.
When flying in Angle Mode with Althold and Poshold activated and start hovering around, everything is fine. But if I try to fly faster and reduce height, the Quadrocopter staggers around with full throttle.
If I switch off Althold and Poshold this effect is gone.
The desaster starts at app. 06:59.4 and ends at 07:04.9

[DIFF_INAV_2.0.txt](https://github.com/iNavFlight/inav/files/2167631/DIFF_INAV_2.0.txt)

[LOG00127.zip](https://github.com/iNavFlight/inav/files/2167632/LOG00127.zip)

https://youtu.be/dbDCWsSmXTE

stronnag, 2018-07-05:
So how does a MR end up in state 29 for 8 seconds?
```
$ inav_modes.rb /tmp/LOG00127.TXT 
iNav version = 2.0.0 (states eq 2.0.0)
Time(s)	Elapsed(s)	State
  87.8	(   0.0)	nav_state_cruise_2d_initialize (29)
  90.7	(   2.9)	nav_state_idle (1)
 107.5	(  19.6)	nav_state_rth_climb_to_safe_alt (9)
 112.0	(  24.2)	nav_state_rth_head_home (10)
 121.7	(  33.8)	nav_state_rth_hover_prior_to_landing (11)
 127.3	(  39.5)	nav_state_cruise_2d_initialize (29)
 128.5	(  40.7)	nav_state_idle (1)
 133.2	(  45.4)	nav_state_althold_in_progress (3)
 135.8	(  48.0)	nav_state_poshold_3d_in_progress (7)
 183.4	(  95.6)	nav_state_althold_in_progress (3)
 183.6	(  95.8)	nav_state_idle (1)
 301.6	( 213.8)	nav_state_althold_in_progress (3)
 301.9	( 214.1)	nav_state_poshold_3d_in_progress (7)
 314.4	( 226.6)	nav_state_althold_in_progress (3)
 314.9	( 227.0)	nav_state_idle (1)
 356.1	( 268.3)	nav_state_rth_head_home (10)
 372.2	( 284.4)	nav_state_idle (1)
 452.7	( 364.9)	nav_state_rth_head_home (10)
 472.8	( 385.0)	nav_state_rth_hover_prior_to_landing (11)
 478.0	( 390.1)	nav_state_cruise_2d_initialize (29)
 486.3	( 398.5)	nav_state_poshold_3d_in_progress (7)
 512.7	( 424.9)	nav_state_althold_in_progress (3)
 512.7	( 424.9)	nav_state_idle (1)
 574.0	( 486.2)	nav_state_althold_in_progress (3)
 574.0	( 486.2)	nav_state_poshold_3d_in_progress (7)
```

shellixyz, 2018-07-05:
> So how does a MR end up in state 29 for 8 seconds?

Very good question. And how its that related to the full throttle since cruise 2D doesn't control the throttle.

Dronitoni, 2018-07-31:
In the meantime I switched from INAV 2.0 RC1 to RC4.
I also changed some paramters in the Configurator.
With the current settings I was not able to provoke this weired behavior again.

So, I will close this issue.

closed 2018-07-31 by Dronitoni