iNavFlight/inav #3525: Feature Request Using the buzzer / beeper as an acoustic Vario (Variometer)

konraddi, 2018-07-06:
Hi dear dev team,

is it theoretically possible to use the buzzer / beeper attached to an iNav build to act as an acoustic vario, beeping when sinking or climbing? Then an onboard micro could listen to the beeps up and send it over the audio-in of the VTX down to the ground into to the goggles.

Or is it otherwise possible to send those beeps directly onto the audio-in on the VTX via some kind of UART / pin of the FC connected via wire to the VTX audio-in?

My aim would be to get the beeps over the air into my goggles audio, using it in thermalling of a ASW 28 v2 fpv glider.

Thanks in advance,
Konrad

teckel12, 2018-07-06:
Not possible using the buzzer because the circuit is only on or off. The buzzer creates the beep frequency on its own. INAV could beep, but can't change the pitch.

If you're using FrSky, you can create a simple set of functions to create vario beeping. You need to be passing telemetry to your transmitter for this to work.

wx4cb, 2018-07-07:
@teckel12 tim, pitch yes. but maybe just have beeps faster/slower depending on RoC/RoD? 

The problem i see with that is cpu usage.

konraddi, 2018-07-07:
Someone on Facebook nentioned this;

"Why not use one of the pwm pins in the fc to generate a tone directly into the audio input of the vTX?
That way you can go further away."

Could that world? It sounds like the perfect solution for this, you would only need one free pwm channel and of course the audio in on the vtx

wx4cb, 2018-07-07:
theoretically yes, but depends on available timers and also cpu usage i would think

r3df1v3, 2018-07-07:
Hi. I've been using this arduino based variometer from Rolf R. Bakke on my FPV EasyGlider Pro:
https://www.rcgroups.com/forums/showthread.php?1749208-DIY-simple-and-inexpensive-Arduino-based-sailplane-variometer
Maybe the code would be useful to any developer implementing a PWM output based solution for iNav. Iwould be great to have the vario integrated with the FC..

teckel12, 2018-07-08:
Honestly, it's so much easier to just let the transmitter do it.

konraddi, 2018-07-23:
How can I do it with my transmitter? Can I use the iNav & crossfire telemetry on my taranis for that?
Thanks, Konrad

teckel12, 2018-07-23:
@konraddi Vario beeping is built into Taranis. I believe it's in the model config on the telemetry page.

closed 2018-10-05 by stale[bot]