iNavFlight/inav #3526: INAV 2.0.0 RC2 doesn't work with FrSky XSR-M S.Port telemetry

teckel12, 2018-07-06:
### Board and Version
Omnibus F4 Pro
INAV/OMNIBUSF4PRO 2.0.0 Jul  6 2018 / 09:28:45 (afc56027)
It's custom compiled, with the only inclusion being #3519, same result with downloaded release RC2 

### Behavior
No S.Port telemetry from INAV from FrSky XSR-M receiver (latest firmware).  Downgraded to INAV 1.9.1 and S.Port telemetry worked again.  Also replaced XSR-M receiver with a backup and it also didn't get telemetry.  Flashed back INAV 2.0.0 RC2 and it still didn't work.  Works fine on other quad running 2.0.0 RC2 and similar hardware except for X4R-SB receiver.

#### Blackbox logs/Config (.ZIP) where problem occurs
https://pastebin.com/xp5iaxHC

teckel12, 2018-07-06:
Got it working, but with a sacrifice.  With 2.0.0 RC2 and the XSR-M/E you need to set `smartport_uart_unidir = ON` for S.Port telemetry to work.  On a different quad with the same firmware, same FC, and almost exactly the same settings, but using an X4R-SB receiver I have `smartport_uart_unidir = OFF`.  Both receivers are connected exactly the same way, to the SBUS port and S.Port connected to soft serial via the ch5/ch6 pads.

I believe what this means is that with INAV 2.0.0 RC, the XSR-M/E (at least when using soft serial) can't do bidirectional S.Port as well as `smartport_uart_unidir = ON` being required.

fiam, 2018-07-07:
@teckel12 I've just set up an XSR (note the different RX) with an OMNIBUSF4PRO and it works with S.Port over softwareserial with `smartport_uart_unidir = OFF`. Could you use `DEBUG_TRACE()` to figure out if the port is being open properly and if the S.Port driver is getting the requests from the RX?

teckel12, 2018-07-08:
@fiam will do... Note that my X4R-SB also works fine with smartport_uart_unidir = OFF. Also note that I'm running S.Port via SS in all cases.

teckel12, 2018-07-08:
@fiam Did a build with DEBUG=GDB and flashed but that's as far as I got.  I was expecting more of log, but it seems that it uses a debugger of some kind?  I've never needed to use a debugger so maybe this is where it's beyond my experience.  I'm old, so I learned to program before we had debugging tools.  I avoid writing bugs, so I've never needed a debugger, I guess till now.

Is this what's needed?: https://gnu-mcu-eclipse.github.io/debug/install/

If so, I could give it a go.  But like I said, I've never used a debugger so my effectiveness would probably be low.

teckel12, 2018-07-09:
:+1: Will give it a go.

teckel12, 2018-08-23:
Now works fine with 2.0.0 on this quad with the `smartport_uart_unidir = OFF`, so closing...

closed 2018-08-23 by teckel12