iNavFlight/inav #3534: iNav 2.0.0RC2 (&RC1) Omnibus F4V5 UART6 failiure

putimir, 2018-07-07:
Hi, I'm having GPS problems (GPS failure - no messages) after upgrading to 2.0.0RC2 on one of my quads. I'm using exact same setup on both versions:

serial 5 2 115200 38400 0 115200

set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_dyn_model = AIR_1G
set gps_auto_config = ON
set gps_auto_baud = ON
set gps_ublox_use_galileo = OFF
set gps_min_sats = 6

If I revert to 1.9.0 or 1.9.1 it works without a hitch

The board is Airbot Omnibus F4V5 and I'm using OB43 target as suggested by Airbot

Any ideas? Could it be something with port defaults for UART6? I think it is meant to be used for serialrx...

X-FOOT, 2018-07-07:
me to. on omnibus F4 pro V2.1  on inav 1.9.1 it works flawless.  and on 2.0 it seems like its not communicating. GPS logo stays RED.  on uart 6, gps settings are: Baudrate: 38400/ M8N/UBLOX/autodetect ground assistance detect.   compass is working fine on uart3

fiam, 2018-07-07:
@X-FOOT your problem is likely not related to this issue. Please, double check you've configured everything correctly. If it still fails, post a config diff.

@putimir Try with the hex attached to this comment and let me know if that works.
[inav_2.0.0_OMNIBUSF4V3_FIX_3534.hex.zip](https://github.com/iNavFlight/inav/files/2172971/inav_2.0.0_OMNIBUSF4V3_FIX_3534.hex.zip)

putimir, 2018-07-07:
OMFG: that was lightning fast 🥇 
I can confirm that fix resolves my issue! 
[image]
Can you be more specific on the background of this issue (just curious) 

Kudos, I take my hat off!

fiam, 2018-07-07:
@putimir It's already fixed in the development branch. PR is at #3485. Long story short: there's an extra inverter in the UART6_TX line in that FC that no one had noticed, but INAV before 2.0 was disabling it by chance. All timers were initialised wether they were used or not and that caused the pin driving the inverter to go low and disable it. Once we removed that needed initialisation, the pin was left it its default state (input with weak pull-up), which caused the inverter to be enabled and invert everything coming out of that TX line.

@X-FOOT Please, open another issue with a config diff if your problem persists. I'm marking this one as fixed.

closed 2018-07-07 by fiam