iNavFlight/inav #3539: No motors show in Motors Tab

GaryGallegos, 2018-07-08:
When I flash iNav onto my Matek F722 FC the motor tab doesn't show any of my motors.  The only thing I see is a master slider.  I used to see all 4 motors but since flashing to Cleanflight and then back to iNav I don't see any motors.
[image]

I AM able to see my motors in Cleanflight and Betaflight however.  Any ideas on what I might be doing wrong?

teckel12, 2018-07-08:
Sounds like your not running 2.0.0 RC2 firmware, or maybe not INAV at all. Go to CLI and do a dump to verify firmware version.

GaryGallegos, 2018-07-08:
Here is the dump:

# dump

# version
# INAV/MATEKF722 1.9.1 Apr 23 2018 / 10:45:17 ()

# resources

# mixer
mixer QUADX

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
[... 501 lines cut ...]
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

teckel12, 2018-07-08:
You're running INAV v1.9.1 with the configurator v2.0.0 RC2.  You need to flash 2.0.0 RC2 if you're using configurator 2.0.0 RC2.

GaryGallegos, 2018-07-08:
Thank you!  It works now.  For some reason I thought 1.91 was the firmware version for the FC only and had nothing to do with iNav version.  Man do I feel foolish, thank you so much!!

closed 2018-07-08 by DzikuVx