iNavFlight/inav #3543: Glitched characters on the OSD

DerEnderKeks, 2018-07-08:
### Board and Version
[Matek F722](https://www.banggood.com/Matek-Systems-F722-STD-STM32F722-Flight-Controller-Built-in-OSD-BMP280-Barometer-Blackbox-for-RC-Drone-p-1225166.html)
INAV 1.9.1

`INAV/MATEKF722 1.9.1 Apr 21 2018 / 13:43:00 (03a5c1922)`

### Behavior
On the OSD some characters are somewhat glitched as visible in [this video](https://www.youtube.com/watch?v=R8xC9PRqmZw).
Mainly the battery voltage (at the top in the middle above the heading graph) and the current draw (on the left) partially show weird symbols instead of the actual values.

I have the OSD set up like this: 
[image]

DzikuVx, 2018-07-08:
Di you uploaded latest fonts?

DerEnderKeks, 2018-07-08:
After uploading the latest font it seems to work fine. But shouldn't the font be included by default?

closed 2018-07-08 by DerEnderKeks