iNavFlight/inav #3545: v2.0.0 RC2 shows GPS altitude -17m before GPS fix

teckel12, 2018-07-08:
Just started with v2.0.0 RC2.  Before GPS fix, the altitude shows as -17 meters.  Once there's a GPS lock, it appears to be correct (but I guess could still be 17 meters off).

```
timestamp,gpsFix,gpsNumSat,gpsLat,gpsLon,gpsAlt,gpsSpeed,gpsGroundCourse
1531075178170,0,0,0,0,-17,0,0
1531075178271,0,0,0,0,-17,0,0
1531075178369,0,0,0,0,-17,0,0
1531075178472,0,0,0,0,-17,0,0
1531075178571,0,0,0,0,-17,0,0
1531075178671,0,0,0,0,-17,0,0
1531075178771,0,0,0,0,-17,0,0
1531075178870,0,0,0,0,-17,0,0
1531075178971,0,0,0,0,-17,0,0
```

It shows as -17m in the configurator and via telemetry.  Flashed 1.9.1 and it shows 0m before GPS fix.

stronnag, 2018-07-08:
https://github.com/iNavFlight/inav/pull/3501

teckel12, 2018-07-08:
I merged in #3501 and it still shows -17m.  This issue isn't for BB recording, it's showing -17m in the configurator and telemetry.  OSD shows GPS altitude of 0.  Didn't confirm if the BB is correct as I'm not skilled with BB recordings and I believe their binary so not easy to quickly review.

stronnag, 2018-07-08:
Post #3501, BB is correct (is was not in rc2). That PR does not affect anything other than BB.

teckel12, 2018-07-08:
I guess that's what I was clarifying, that #3501 doesn't address this problem.

fiam, 2018-07-11:
@teckel12 I can't reproduce this. The only way I see a -17 would get in there would be if the GPS was sending -17m as its altitude. Note that telemetry and the configurator show the altitude as reported by the GPS, while the OSD shows the estimated altitude (which stays at zero while disarmed).

teckel12, 2018-07-11:
I have 2 quads with the BN-880 GPS. Running 1.9.1 they both show 0m for the GPS altitude. But when I upgraded to 2.0.0 both show - 17m. On one I even rolled back to 1.9.1 and it showed 0m again.

stronnag, 2018-07-11:
In gpspassthrough I see 0N,0E -17m until a position is determined (without a fix)
```
POSLLH: lat: 0.000000 lon: 0.000000 elev: -17.00 acc(h/v): 4294967.3/3750000.4
sats: 0, fix 0
```
then as it started to get a position:
```
POSLLH: lat: 26.415727 lon: 6.939176 elev: 4174.57 acc(h/v): 4294967.3/3750001.9
sats: 0, fix 0
```
This is an old bench test Neo6M, doesn't get a fix very often, today is its lucky day.
```
POSLLH: lat: 50.915381 lon: -1.530285 elev: 12.00 acc(h/v): 17.1/20.3
sats: 5, fix 3
```
Noted that gps_ublox.c has not changed since 1.9.0 ...

But so what, if your GPS has no fix, it doesn't really matter what the altitude is. Let me find a 1.9 hex.

fiam, 2018-07-11:
I've just realised why you don't see that in INAV 1.9.1. It's because the GPS wasn't sent until there was a GPS fix. Now it's sent either if you haven't armed or if it has a fix. This way you can discover the sensors without having to get a fix first, but if the GPS stops working in flight you will notice.

closed 2018-07-11 by fiam

teckel12, 2018-07-11:
@fiam That makes sense!  Doesn't make much sense that GPS receivers are outputting a value of -17m before there's a GPS lock, but not INAV's fault.