iNavFlight/inav #3549: Disarm mid-air with iNav 2.0 (b569c3051) after activating launch mode

flounderscore, 2018-07-09:
### Board and Version
INAV/OMNIBUSF4PRO 2.0.0 Jul  6 2018 / 21:46:22 (b569c3051)

### Behavior
I recently upgraded from 1.7 to the latest GitHub version (full chip erase and manual reconfiguration). 

First two flights with the new iNav on my MiniDrak  went fine without any issues (see BB_SUCCESS.zip for the second one). After that I decided to activate the launch mode which I'm using on another wing (w/o GPS, iNav master from a few days ago) without any issues. Seconds after I took control after a successful launch, the FC disarmed mid-air showing disarm reason "Switch" on the OSD. I tried to toggle the arming switch, but at that moment I forgot that the launch mode was still activated... Luckily, I still had control and landed dead-stick.

At that point, I was convinced I must have accidentally bumped the arm switch and gave it another try. Same behavior, a few seconds after the launch the FC disarmed, but this time I had no control at all and crash-landed in the trees.

The XSR RX I'm using always shows RSSI lower than another R-XSR on the second plane, but I haven't had any range issues in the past at the field I'm flying (never with 1.7 and 2 successful flights with 2.0 w/o launch mode). Maybe it is related to #2981 or maybe I just screwed up my configuration in some weird way (disarm when auto launch times out?) Maybe you guys can see what's wrong in the logs/config.

#### Blackbox logs/Config (.ZIP) where problem occurs
[BB_SUCCESS.zip](https://github.com/iNavFlight/inav/files/2174361/BB_SUCCESS.zip)
[BB_DISARM.zip](https://github.com/iNavFlight/inav/files/2174360/BB_DISARM.zip)

[DIFF.TXT](https://github.com/iNavFlight/inav/files/2174362/DIFF.TXT)

teckel12, 2018-07-09:
Possible #3482

flounderscore, 2018-07-09:
I agree, but if that's was the only issue I cannot explain the failure rates: w/o launch mode 0 failures/10 minutes and then later w/ launch mode 2 failures/10seconds.

digitalentity, 2018-07-16:
Since https://github.com/iNavFlight/inav/pull/3482 is merged, RC receiver failsafe settings won't mess up processed channel values (`rcData`). I think this issue should now be fixed.

flounderscore, 2018-07-16:
When I have time, I'll tried to reproduce my issue (while waking the wing) and then upgrade to the latest version.

shellixyz, 2018-07-22:
@flounderscore If you are using the launch feature you can regain the control of the motor after a mid air disarm by moving the roll/pitch stick while the throttle is still at 0 after rearming.

digitalentity, 2018-07-29:
@flounderscore any update?

flounderscore, 2018-07-30:
@digitalentity Unfortunately, not yet. I haven't found time to fly again. I'll report back as so as possible.