iNavFlight/inav #3552: QUAD The motors spins independently of the value of the check "dont spin ..."

fmartinezo, 2018-07-09:
_Board Matek 405 ctr
_Version of INAV 2.0.0 RC2

### Behavior
*The motors spin independently of the value of the check "dont spin ..."
* Firm was uploaded with full chip erase. 
* _Expected Results_ Mark check "dont spin....." it lights blue, arm quad and should not spin
* _Actual Results_Mark check "dont spin....." it lights blue, arm quad and motors begin spin

_Is this a new feature request?_ no

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.
# diff

# version
# INAV/MATEKF405 2.0.0 Jun 26 2018 / 22:47:39 (5db1e1a45)

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature -TX_PROF_SEL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature LED_STRIP
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 2 2 115200 38400 0 115200
serial 30 32 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 1700 2100
aux 1 1 0 1300 2100
aux 2 2 0 900 1250
aux 3 3 0 1700 2100
aux 4 9 0 1700 2100
aux 5 8 7 1700 2100
aux 6 11 6 1700 2100
aux 7 12 3 1700 2100
aux 8 15 1 1700 2100

# adjrange

# rxrange

# osd_layout

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set acc_hardware = MPU6000
set acczero_x = 69
set acczero_y = -14
set acczero_z = -145
set accgain_x = 4084
set accgain_y = 4098
set accgain_z = 3978
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 45
set magzero_y = -121
set magzero_z = -50
set baro_hardware = BMP280
set pitot_hardware = NONE
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set model_preview_type = 3
set nav_disarm_on_landing = ON
set nav_user_control_mode = CRUISE
set nav_auto_speed = 500
set nav_manual_speed = 800
set nav_mc_bank_angle = 40
set name = SMAC300

# profile
profile 1

set mc_i_pitch = 12
set mc_d_pitch = 22
set mc_i_roll = 12
set mc_d_roll = 22
set mc_p_yaw = 50
set mc_i_yaw = 12
set mc_d_yaw = 22
set mc_p_level = 35
set max_angle_inclination_rll = 400
set max_angle_inclination_pit = 400
set dterm_lpf_hz = 80
set dterm_notch_hz = 200
set dterm_notch_cutoff = 100
set rc_expo = 20
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45

# battery_profile
battery_profile 1

# 

_Upload a zipped blackbox log (if available)_

giacomo892, 2018-07-09:
Hi. Have you calibrated the ESC?

fmartinezo, 2018-07-09:
Yes, four esc was calibrated. I don't have this problem on inav 1.8 or 1.9

MRC3742, 2018-07-12:
Same problem but a little more details. 
Don't spin is selected in configurator and works as expected if I arm in Angle Mode.
Arming or changing to Acro or Horizon Mode cause the motors spin up.

SP Racing F3 EVO
iNav 2.0.0_RC2

DzikuVx, 2018-07-13:
In INAV 2.0 Airmode overrides MOTOR_STOP feature. When Airmode is enabled, motors will always spin since those 2 features excludes themselves on multirotors

closed 2018-07-13 by DzikuVx