iNavFlight/inav #3554: inav 2.0.0-RC2 Navigation not safe: Why? After unplugging the USB cable the motors do not start, too.

Ralfde, 2018-07-09:
Omnibus F4V5 with F4V3 firmware
inav 2.0.0-RC2

[image]

[image]

[image]

Entering CLI Mode, type 'exit' to return, or 'help' # dump

# version
# INAV/OMNIBUSF4V3 2.0.0 Jul 8 2018 / 21:49:00 (a8415e89)

# resources

# mixer

mmix reset

mmix 0 1.000 -1.000 -1.000 -1.000
mmix 1 1.000 1.000 1.000 -1.000
mmix 2 1.000 1.000 -1.000 1.000
mmix 3 1.000 -1.000 1.000 1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 
servo 1 1000 2000 1500 100 
servo 2 1000 2000 1500 100 
servo 3 1000 2000 1500 100 
servo 4 1000 2000 1500 100 
servo 5 1000 2000 1500 100 
servo 6 1000 2000 1500 100 
servo 7 1000 2000 1500 100 

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
[... 857 lines cut ...]
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 70
set roll_rate = 77
set pitch_rate = 77
set yaw_rate = 70
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 1300
set battery_capacity_warning = 195
set battery_capacity_critical = 130
set battery_capacity_unit = MAH

Ralfde, 2018-07-09:
After updating again, problem seem to be gone.

stronnag, 2018-07-09:
https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons

closed 2018-07-09 by Ralfde