iNavFlight/inav #3556: Huge lag in heading calculation - airplane/GPS MTK only 1.9.1/2.0

marbalon, 2018-07-09:
Today I tried to fly and tune my Glider with GPS MTK GlobalTop Titan 3, GMS-G9, GPS/GLONASS, MT3333. I'm using this GPS in this model from the beginning. I made a lot of flight on 1.7.1 and navigation works realy good. 

But When I upgrade this model to 1.9.1 I have a lot of problem with navigation mode because model overshot home direction but that was not because PID but because lag in course calculation. Then I have no time to try it more on 1.9.1 and today I uploaded v2.0 (compiled yesterday from development branch), but navigation problem is the same - huge overshot.

So my question is what changes between this version can bring this problem ? Something was changed in heading calculation in plane with GPS only (no MAG)? Can I tune FC option to make it more similar to 1.7.1 ?

diff:
`Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/KFC32F3_INAV 2.0.0 Jul  8 2018 / 19:38:08 (5ef4fb9)

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix 0 0 2 100 0
smix 1 1 1 100 0
smix 2 2 4 100 0
smix 3 3 11 100 0
smix 4 4 4 -100 0
smix 5 5 11 -100 0
smix 6 6 9 100 0
smix 7 7 10 100 0

# servo
servo 0 1000 2000 1500 95
servo 1 1000 2000 1500 -70
servo 2 1000 1800 1499 -90
servo 3 1000 1750 1598 -100
servo 4 1175 1900 1423 90
servo 5 1000 2000 1124 -100
servo 6 750 2220 1521 125
servo 7 800 1950 1500 125

# feature
feature -TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 1 2 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1525 1700
aux 1 1 7 900 1300
aux 2 3 0 1450 1575
aux 3 3 5 900 1300
aux 4 9 0 1450 1575
aux 5 8 0 1700 1800
aux 6 35 5 900 1300
aux 7 10 7 1700 2100

# adjrange

# rxrange

# osd_layout
osd_layout 0 1 16 0 V
osd_layout 0 4 8 6 H
osd_layout 0 6 22 12 V
osd_layout 0 7 12 13 V
osd_layout 0 9 23 1 V
osd_layout 0 11 1 1 V
osd_layout 0 12 16 1 V
osd_layout 0 13 2 7 V
osd_layout 0 14 0 12 V
osd_layout 0 20 18 13 V
osd_layout 0 21 0 13 V
osd_layout 0 22 14 12 V
osd_layout 0 23 7 0 V
osd_layout 0 24 7 1 V
osd_layout 0 25 23 5 V
osd_layout 0 26 24 7 V
osd_layout 0 28 23 11 H
osd_layout 0 30 2 14 V
osd_layout 0 41 1 9 H
osd_layout 0 42 1 10 H
osd_layout 0 62 0 10 H
osd_layout 0 63 0 10 H

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6000
set acczero_x = 42
set acczero_y = -3
set acczero_z = -74
set accgain_x = 4068
set accgain_y = 4065
set accgain_z = 4078
set mag_hardware = NONE
set baro_hardware = MS5611
set pitot_hardware = NONE
set receiver_type = SPI
set rx_spi_protocol = ELERES
set blackbox_rate_denom = 16
set min_throttle = 1090
set failsafe_procedure = RTH
set align_board_pitch = -34
set vbat_scale = 1115
set current_meter_scale = -110
set current_meter_offset = 8075
set platform_type = AIRPLANE
set model_preview_type = 14
set servo_init_after_gyro = ON
set small_angle = 180
set disarm_kill_switch = OFF
set gps_provider = NMEA
set inav_reset_home = FIRST_ARM
set inav_w_z_baro_p =  1.000
set inav_w_z_gps_p =  0.000
set inav_w_z_gps_v =  0.000
set nav_extra_arming_safety = OFF
set nav_landing_speed = 100
set nav_land_slowdown_minalt = 900
set nav_rth_altitude = 10000
set nav_fw_cruise_thr = 1350
set nav_fw_max_thr = 1500
set nav_fw_bank_angle = 15
set nav_fw_climb_angle = 10
set nav_fw_dive_angle = 5
set nav_fw_pitch2thr = 20
set nav_fw_loiter_radius = 7000
set eleres_telemetry_en = ON
set eleres_loc_en = ON
set eleres_signature = 3353758710
set osd_video_system = PAL
set osd_time_alarm = 500
set osd_alt_alarm = 1500
set osd_dist_alarm = 10000
set async_mode = GYRO

# profile
profile 1

set fw_p_pitch = 20
set fw_i_pitch = 43
set fw_ff_pitch = 200
set fw_p_roll = 20
set fw_i_roll = 43
set fw_ff_roll = 200
set fw_p_yaw = 19
set fw_i_yaw = 42
set fw_ff_yaw = 193
set fw_p_level = 50
set fw_i_level = 15
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 450
set nav_fw_pos_z_p = 30
set nav_fw_pos_xy_p = 30
set nav_fw_pos_xy_d = 12
set roll_rate = 9
set pitch_rate = 6
set yaw_rate = 3

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 300`

GareFPV, 2018-07-10:
Same problem after upgrading from 1.8 to 2.0, GPS overshot home direction :-( Having a huge lag also.

GareFPV, 2018-07-21:
I deactivated the barometer and after that the GPS lag was gone...

digitalentity, 2018-07-22:
@GareFPV barometer doesn't affect heading calculation. Blackbox logs with & without the issue (as well as config dumps for both cases) would be mandatory to help debug.

shellixyz, 2018-08-03:
I encountered the issue and a workaround seems to be to disable SBAS.

giacomo892, 2018-09-10:
@digitalentity could be nice if we change the auto-configuration process to also check if the settings were actually set as intended and if not, apply them again.

ShikOfTheRa, 2018-09-11:
I have this issue. It's on 2.0 with first test on a new build on weekend. But I have also experienced in 1.9 
I have just prepped a new and different GPS , but might be another 2 weeks.

its a ublox (likely a clone) running 38400 with default inav settings on F405wing Cpu is very low.. In all other ways its seems to work - lots of sats, several osd updates per second.
If I land it takes 4-8 secs to achieve steady state. Any nav funtionality / home arrow is of no value.

Can do bench testing it it helps, but probably 2 weeks before flying again unless am lucky...

Can show a flight vid where its very visible