iNavFlight/inav #3560: New Feature to set VTX Power only has 4 Levels. IRC Tramp has 5 Levels of Power

haksarion, 2018-07-10:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Furious FPV F35 Lightning FC

### New VTX Power Feature only has 4 Levels
My IRC Tramp is connected to the F35 via Telemetry Port. In OSD I can easily change channels and VTX Power.
The new Feature to set VTX Power on Arming and Disarming works brilliant, except one little thing.
The IRC Tramp has 5 Power Levels... but Inav only supports 4 Power Levels.
So if I want to set it to highest possible output Power I set "set vtx_power = 5" but only Values from 1 to 4 are allowed.

_Is this a new feature request?_ YES

Greetz

fiam, 2018-07-10:
@haksarion There are 5 available power levels, from 0 to 4. That allows selecting 5 different power levels.

haksarion, 2018-07-11:
RC-Tramp:  0=25mW, 1=25mW, 2=100mW, 3=200mW, 4=400mW, 5=600mW

GareFPV, 2018-07-11:
for SmartAudio:  0=25mW, 1=25mW, 2=200mW, 3=500mW, 4=800mW
for TBS Unify Nano:  0=25mW, 1=25mW, 2=50mW
for IRC-Tramp:  0=25mW, 1=25mW, 2=100mW, 3=200mW, 4=400mW, 5=600mW

fiam, 2018-07-11:
Looks like the problem is that the levels are shifted down by 1 somewhere.

fiam, 2018-07-11:
Fixed in #3564, will be in RC3

commit 13699d29 (2018-07-11): Fix VTX power levels to be in 1-5 range rather than 0-4

Drivers ignore zero, so the minimum power is actually 1.
Also, remove some hardcoded values and use proper constants
for them, fixing an issue with the number of power levels in
Tramp (sizeof of an uint16_t array was used rather than its count).

Fixes #3560
commit be2dd33d (2018-07-11): Fix VTX power levels to be in 1-5 range rather than 0-4

Drivers ignore zero, so the minimum power is actually 1.
Also, remove some hardcoded values and use proper constants
for them, fixing an issue with the number of power levels in
Tramp (sizeof of an uint16_t array was used rather than its count).

Fixes #3560
PR #3564 (2018-07-11): Fix VTX power levels to be in 1-5 range rather than 0-4
Drivers ignore zero, so the minimum power is actually 1.
Also, remove some hardcoded values and use proper constants
for them, fixing an issue with the number of power levels in
Tramp (sizeof of an uint16_t array was used rather than its count).

Fixes #3560
closed 2018-07-11 by fiam