iNavFlight/inav #3563: v2.0.0 RC2 Intermittent SBUS connection on Matek F405-STD and RadioLink R9DS

Andreas-Hellberg, 2018-07-11:
### Board and Version
Matek F405-STD BetaFlight STM32F405 Flight Controller 
https://www.banggood.com/Matek-F405-OSD-BetaFlight-STM32F405-Flight-Controller-Built-in-OSD-Inverter-for-RC-Multirotor-FPV-Racing-Drone-p-1141282.html?rmmds=search
v2.0.0 RC2
INAV/MATEKF405 2.0.0 Jun 29 2018 / 12:30:25 (a8415e89c)

### Behavior
When I tested upgrading to INAV 2.0 RC2 the SBUS connection became intermittent.
The board will read the stick positions for 0.5-1s then there will be a period of no changes for 1-2s then it reads the current RC input for 0.5s-1s and repeat. It's the same result when connected via USB and disconnected.

The same hardware works just fine on all previous versions of INAV I have tested (1.9.3, 1.9.0, etc). 
I flashed the board with the "full chip erase" option set to true. I tested going back an forth between 1.9 and 2.0 multiple times. On 1.9 it works just fine, but not on 2.0. 

#### Blackbox logs/Config (.ZIP) where problem occurs
https://pastebin.com/Sypn6ReH

giacomo892, 2018-07-11:
Perhaps the receiver isn't respecting SBUS timings. 
Can you set debug to `SBUS`and check what happening on the configurator Sensors tab in the debug data?

Andreas-Hellberg, 2018-07-11:
debug_mode set to SBUS: 
[image]

How it looks in the receiver tab:
[image]

giacomo892, 2018-07-11:
Seems like the DESYNC_COUNTER increases... then it is an issue the the receiver not respecting SBUS specifications.

Andreas-Hellberg, 2018-07-11:
Might be, but the receiver works fine in the previous releases with the old SBUS driver.

giacomo892, 2018-07-11:
Older INAV versions were much more relaxed about SBUS input.
In 2.0 we have introduced a more strict approach when decoding SBUS packet to be sure that all the data is decoded correctly and no channels have invalid data.

Andreas-Hellberg, 2018-07-11:
To me this is a regression rather than an improvement. But hey I'm a bit biased as this stops me from upgrading.. :) I know I could create my own release and just revert the SBUS commits. Might do that in the future if this issue is closed without an resolution. But it could be that the new SBUS driver excludes more than just RadioLink receivers and that would be less than ideal if so.

digitalentity, 2018-07-11:
None of the devs has a RadioLink system, so troubleshooting this might be problematic.

digitalentity, 2018-07-11:
@giacomo892 @fiam I suggest to add this to known issues section and look into it after 2.0 release.

giacomo892, 2018-07-11:
Added to wiki @digitalentity

Andreas-Hellberg, 2018-07-11:
I tested reverting PR 3110 (https://github.com/iNavFlight/inav/pull/3110/commits) and built + flashed it. 
That worked so that will have to be my workaround for the time being. 

[image]

giacomo892, 2018-07-12:
@Andreas-Hellberg bare in mind you may encounter mid air disarms and any other nasty things.

Andreas-Hellberg, 2018-07-12:
@giacomo892 I don't see why. As the SBUS code is the same in earlier versions of Inav. But even if you are right I think my frequent crashing tells me that potential software bugs is the least of my problems. :) 

I'll probably have to replace my old RadioLink radio with a FrSky sooner rather than later. But my RadioLink gear have worked fine with, Cleanflight, Betaflight, all previos version of Inav and propitiatory software/hardware from DJI and Tarot. So I still think there is something wrong with the new SBUS driver.

giacomo892, 2018-07-15:
@Andreas-Hellberg give a shot to RC4 and ply with the sync delay until you see no sync issues on debug data.

Andreas-Hellberg, 2018-07-15:
@giacomo892 thanks for the reply, I tried changing the SBUS delay from 500,1500,2500 etc up to 10000. Did not see a significant improvement though. The min/max range for the new settings is reversed so I had to fix that before I could test it out. 

[image]

giacomo892, 2018-07-15:
@Andreas-Hellberg could you PR the fix?

giacomo892, 2018-07-15:
You need to Fork, commit to your branch and then PR to inav :)

fiam, 2018-07-15:
@Andreas-Hellberg do you have an oscilloscope to see the frequency of the SBUS signal output by that RX?

Andreas-Hellberg, 2018-07-15:
@giacomo892 thanks for the help. The tiny PR is now up at https://github.com/iNavFlight/inav/pull/3606 :)
@fiam No I don't have access to a oscilloscope. Any way to log the output via a FC?

Andreas-Hellberg, 2018-07-16:
@gppsoft is it just on 1.9.1 or is it all versions? It's a common issue if you have some external component (gps?)  connected to the same UART (TX1/RX1) as the USB port. Then that component will go active once the battery is plugged in an interfere with the USB connection.

rollanc, 2018-07-20:
Tested inav 2.0 rc4 with radiolink receiver R12DS, bus F4 v3, and sbus does not work for auxiliary channels.

digitalentity, 2018-07-22:
@rollanc if it works for main channels, but doesn't for AUX - it's likely a radio setup or messed up configuration issue.

rollanc, 2018-07-23:
No, I have everything perfectly configured, I've already come from Inav 1.6. It must be error in the radiolink ibus frame, why fryski, if that works.

happyvalley-lmx, 2018-10-12:
I'm also using RadioLink R9DS.
Sbus seems work fine on my omnibus F4V3.

JVN2000, 2018-11-11:
I am using Radioink R9DS on a OMNIBUS F4PRO V2 and OMNIBUS F3 running inav 2.0, no problems at all.

digitalentity, 2018-11-11:
Seems this issue is not relevant anymore. Closing.

closed 2018-11-11 by digitalentity