iNavFlight/inav #3574: Inav 2.0 Omnibus f4 pro v2 smartaudio not working

Harald86, 2018-07-12:
I have a omnibus f4 pro v2 and a eachine tx5258 with smart audio, I don't get smartaudio to working, I connected on the smal solder pads ch5 6 near mcu and enabled soft serial, tryed uart 3 and 6 no communication is possible

The omnibus is a clone from banggood, i used different hex files every not works

I have another board open pilot cc3d revo and connected on main port TX and it's working correct!

deef107, 2018-07-17:
I believe uart 3 and 6 are already available on the omnibusF4 pro v2 on the J10 connector. If you enable soft serial you should get other uarts available in the configurator. Did you do resource reallocation? I've connected smartaudio to the j10 connector on tx 3 (or 6) and it worked.

Harald86, 2018-07-17:
I used uart 3 and 6 for gps and Mag so I need more ports and I enabled soft serial for ch 5 6 near mcu

I think inav does not have Ressource remapping, only betaflight does have implemented!

The crazy thing is that is smart audio with the eachine tx5258 and the cc3d revo works and I don't have any chance on the omnibus f4 pro v2 uart 3 or 6 or ch5 ch6

So there must be a software issue or compatibility issue

What Vtx are you using? Tbs?

deef107, 2018-07-17:
I already thought you were maybe using uart 3 and 6 for gps and mag. I don't think you can use ch5 or 6 pins by just enabling soft serial. I think the pins are connected to the existing uart 3+6 tx pins. I didn't know that inav couldn't do resource allocation. I'm wondering what soft serial does if you have no spare pins..? I read somewhere that you can try some custom hexes to get different allocation, but I could understand if you won't like to go that route. I'm using IRC TrampHV so I was incorrect about calling it smart audio, but it should work with both. I presume you have connected your receiver to uart 1, but how? Could it be possible to use tx of uart 1? Outside of the box thinking.... ;)

deef107, 2018-07-17:
I believe it's needed for frsky, but I'm not an frsky user. I've also read on some forums in the meantime that you can't have the receiver and smart audio on the same uart, but I can't verify.
ch5 and 6 won't work' I'm pretty sure, because you are already using it.
I guess you're using inav for multirotor and therefore need the mag?
I had a similar challenge when I wanted an oled display on my omnibus F4 pro V2 and ran out of uarts. Luckily I had chosen the TrampHV and it has this nifty nfc capability to easily configure your vtx with your smartphone. This was my workaround. Next time I'll get an fc with more uarts.

Harald86, 2018-07-17:
Ch5 and 6 is free, I checked with command resource!

I can try connect smartaudio on uart 1 and sbus on ch6 TX

deef107, 2018-07-17:
You could but I think uart6 tx is used by your gps via j10 connector.... ch6 and tx6 pins are connected, I think.

sq9fk, 2018-09-07:
Hello,
Follow is my input in developing Inav 2.0.0 for Omnibus F4 PRO V2 from Banggood.
I've tested following configuration:
I have version where CH5 and CH6 pads are visible and no jumper for SBUS/PPM configuration.

My setup:
UART1 RX - SerialRX (S.bus) (3 pin 2.54)
UART1 TX - SoftSerial2 RX/TX - SmartAudio (4 pin JST 1.0)
UART3 TX/RX - Magnetometr I2C (6 pin JST 1.0)
UART6 TX/RX - GPS (6 Pin JST 1.0)
CH5/CH6 - SoftSerial1 RX/TX - Telemetry (Solder into the pads)

SerialRX, SmartAudio and Telemetry works. Need to change following files in INAV 2.0.0 source:

inav/src/main/target/OMNIBUSF4/target.h: from line 138

```
//Commneted out beacause OMNIBUSF4PRO V2 doesn't have pin for inverter.
//#define USE_UART_INVERTER

#define USE_UART1
#define UART1_RX_PIN            PA10
#define UART1_TX_PIN            NONE // Free UART1_TX pin for SoftSerial1 for ex. SmartAudio
#define UART1_AHB1_PERIPHERALS  RCC_AHB1Periph_DMA2
#if !defined(OMNIBUSF4V3)
//Commneted out beacause OMNIBUSF4PRO V2 doesn't have pin for inverter.
//#define INVERTER_PIN_UART1_RX PC0 // PC0 has never been used as inverter control on genuine OMNIBUS F4 variants, but leave it as is since some clones actually implement it.
#endif

#define USE_UART3
#define UART3_RX_PIN            PB11
#define UART3_TX_PIN            PB10

#define USE_UART6
#define UART6_RX_PIN            PC7
#define UART6_TX_PIN            PC6
#if defined(OMNIBUSF4V3)
  #define INVERTER_PIN_UART6_RX PC8
  #define INVERTER_PIN_UART6_TX PC9
#endif

#if defined(OMNIBUSF4V3)
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_RX_PIN     PC6 //shared with UART6_TX
#define SOFTSERIAL_1_TX_PIN     PC6 //shared with UART6_TX

#define SERIAL_PORT_COUNT       5 //VCP, USART1, USART3, USART6, SOFTSERIAL1
#else
// CH5 / CH6 for ex. Telemetry
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_RX_PIN     PC8
#define SOFTSERIAL_1_TX_PIN     PC9
// Add second SoftSerial
#define USE_SOFTSERIAL2
#define SOFTSERIAL_2_RX_PIN     PA9 // UART1 TX Pin for ex SmartAudio bidir
#define SOFTSERIAL_2_TX_PIN     PA9 // UART1 TX Pin for ex SmartAudio bidir

#define SERIAL_PORT_COUNT       6 //VCP, USART1, USART3, USART6, SOFTSERIAL1, added SOFTSERIAL2
#endif
```

inav/src/main/target/OMNIBUSF4/target.c: from line 48:

```
{ TIM1,  IO_TAG(PA8),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_6
//Add timer for PA9 SoftSerial2 TX
{ TIM1,  IO_TAG(PA9),  TIM_Channel_2, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_ANY }, // SoftSerial2 UART1 TX
```

Hope we can share it to all the issues with SoftSerials and OMNIBUSF4PRO V2 from BG or Ali

closed 2018-11-20 by stale[bot]