iNavFlight/inav #3577: 2.0.0-rc2 - OSD - Flight timer not working

moggiex, 2018-07-12:
Howdy,

The flight timer in this version doesn't appear to be working. 

Can be seen in this video [https://youtu.be/vnWua-0XvKw](https://youtu.be/vnWua-0XvKw) at 4 mins 21 for the launch.

PS. Flew fantastically smoothly, took the laptop out for the maiden and it wasn't needed, YAY!

diff results:

`Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/MATEKF722 2.0.0 Jun 29 2018 / 12:33:01 (a8415e89c)

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 4 0 -50 0
smix 1 4 1 -50 0
smix 2 3 0 -51 0
smix 3 3 1 50 0

# servo

# feature
feature -TELEMETRY
feature MOTOR_STOP
feature GPS
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 1 64 115200 115200 0 115200
serial 2 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1775 2100
aux 1 2 1 1300 1700
aux 2 8 2 1750 2100
aux 3 35 2 1300 1700
aux 4 10 1 900 1275
aux 5 27 3 1325 1725
aux 6 28 3 1700 2100
aux 7 32 4 1750 2100

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 1 1 V
osd_layout 0 1 23 1 V
osd_layout 0 7 1 4 V
osd_layout 0 9 23 3 H
osd_layout 0 11 23 2 V
osd_layout 0 12 23 3 V
osd_layout 0 13 2 7 V
osd_layout 0 14 1 2 V
osd_layout 0 15 23 7 V
osd_layout 0 20 18 14 V
osd_layout 0 21 1 14 V
osd_layout 0 22 11 1 V
osd_layout 0 23 13 1 V
osd_layout 0 28 1 3 V
osd_layout 0 33 23 4 V
osd_layout 0 34 10 12 V
osd_layout 0 40 23 11 V
osd_layout 0 48 1 3 H
osd_layout 1 0 1 1 V
osd_layout 1 1 23 1 V
osd_layout 1 3 8 6 V
osd_layout 1 7 1 4 V
osd_layout 1 11 23 2 V
osd_layout 1 12 23 3 V
osd_layout 1 13 1 7 V
osd_layout 1 14 1 2 V
osd_layout 1 15 24 7 V
osd_layout 1 20 18 12 V
osd_layout 1 21 1 12 V
osd_layout 1 22 11 1 V
osd_layout 1 23 13 1 V
osd_layout 1 28 1 3 V
osd_layout 1 30 1 13 V
osd_layout 1 33 23 4 V
osd_layout 1 34 10 11 V
osd_layout 1 38 24 2 H
osd_layout 1 45 0 0 V

# master
set looptime = 2000
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 40
set acc_hardware = MPU6500
set acczero_x = -28
set acczero_y = -13
set acczero_z = 14
set accgain_x = 4090
set accgain_y = 4083
set accgain_z = 4093
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 12
set failsafe_procedure = RTH
set vbat_scale = 1093
set platform_type = AIRPLANE
set model_preview_type = 8
set small_angle = 180
set gps_sbas_mode = AUTO
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 7500
set nav_fw_bank_angle = 30
set nav_fw_climb_angle = 30
set nav_fw_loiter_radius = 6000
set osd_units = UK
set osd_time_alarm = 20
set osd_alt_alarm = 500
set osd_dist_alarm = 4000

# profile
profile 1

set fw_p_pitch = 6
set fw_i_pitch = 9
set fw_ff_pitch = 52
set fw_p_roll = 6
set fw_i_roll = 6
set fw_ff_roll = 49
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set tpa_rate = 20
set tpa_breakpoint = 1600
set roll_rate = 28
set pitch_rate = 14

# battery_profile
battery_profile 1

set battery_capacity = 3000
set battery_capacity_warning = 2100
set battery_capacity_critical = 2700

# `
Matt

moggiex, 2018-07-12:
Had a look at the version history, is this related?

[https://github.com/iNavFlight/inav/commit/f1c8ac484a7775c45389a023bdad96179944becc](https://github.com/iNavFlight/inav/commit/f1c8ac484a7775c45389a023bdad96179944becc)

Matt

giacomo892, 2018-07-12:
@moggiex Yes it is.
Should be fixed in the freshly released RC3. Go to upgrade and report! Thanks!

moggiex, 2018-07-12:
Cheers, will load and test tomorrow & report back,

Matt

moggiex, 2018-07-13:
Yes, fixed in RC3

closed 2018-07-13 by moggiex