iNavFlight/inav #3578: New feature request - Flight Path Vector

wojciechwas, 2018-07-12:
Could you consider adding this feature please? 

Flight Path Vector is a circular symbol that displays estimated drift angle and flight path angle. 

When shown on OSD it can be very useful instrument to assist on landing approach, general flying and hitting gaps.... 

More info on the flight path vector: 
http://www.flaps2approach.com/journal/2012/7/7/flight-path-vector-fpv-explanation-and-use.html 
http://www.smartcockpit.com/docs/Fokker_100-Flight_Path_Vector_Use.pdf 
https://www.youtube.com/watch?v=wSR0L7FGM4o

fiam, 2018-07-12:
@wojciechwas That's definitely interesting, but I'm not sure if we can pull it off with the drawing resolution we have in the OSD (12x18 character grid, characters have to be aligned to it).

wojciechwas, 2018-07-12:
Could it not be displayed in the same way as the artificial horizon? 
The flight path vector circle can be rendered as a bunch of white dots instead if a special character. 
Therefore the flight-path vector can be allowed to float in the middle of the screen just as the AH, without character grid constraint.

wojciechwas, 2018-08-02:
More links on how to implement flight pat vector: 
https://diydrones.com/profiles/blogs/airspeed-angle-of-attack-sideslip-and-flight-path-vector 
https://people.mpi-sws.org/~bbb/proceedings/rtas14-wip-proceedings.pdf

teckel12, 2018-08-02:
@wojciechwas I believe with magnatometer data I could calculate the flight path vector in Lua Telemetry on the new pilot view (glass cockpit).  It wouldn't be heads up, but would show on your transmitter.

Do you fly Taranis/SmartPort?

wojciechwas, 2018-08-03:
@teckel12 That would be a good to have feature even on the glass cockpit. 
Would it be fairly straight forward for developers to pull this into the OSD at later stage? 
I manly fly on googles and having the flight path vector on the OSD would be more useful. 
Would be happy to try it out with the pilot view first and so some instrument flying. 
Yes I use X9E and Smartport telemetry.

teckel12, 2018-08-03:
The problem with on the OSD is the limited "resolution" as everything on the display is just a character which is mapped to fixed locations.  There also can't be two characters in the same spot or one will cover the other or cause flashing.  Basically, the OSD isn't pixel based, it's character based.  This is why the artificial horizon on the OSD is kinda janky.  On the transmitter we have pixel-level control so this isn't a problem.

closed 2023-03-06 by DzikuVx