iNavFlight/inav #3579: inav 2.0.0 rc3 max bank angle half of what expected.

lawnmower4x, 2018-07-12:
### Board and Version
_ Omnibus corner nano. 
https://www.banggood.com/Omnibus-F4-Corner-Nano-Flight-Controller-Build-In-OSD-Current-Sensor-With-Damping-Box-IMU-p-1231552.html?rmmds=search&cur_warehouse=CN
_ inav_2.0.0-rc3_OMNIBUSF4V3.hex

_Use CLI command `version` and put its output here_
# INAV/OMNIBUSF4V3 2.0.0 Jul 12 2018 / 11:35:21 (81ce75745)

### Behavior
* In Advanced tuning tab I set the "Multirotor max. banking angle" to the maximum 45 degrees, I expect the quad to tilt 45 degrees in all directions in angle mode.
The quad only tilts around 20 degrees ( or 22.5?) in all degrees.

The flight controller is, as far as I know, equipped with a ICM20608 gyro/accellerometer. inav reports it having an mpu6500 sensor.

How can I make the quad tilt 45 degrees in all directions in angle mode?

#### Blackbox logs/Config (.ZIP) where problem occurs
Got no blackbox, config is:

# dump

# version
# INAV/OMNIBUSF4V3 2.0.0 Jul 12 2018 / 11:35:21 (81ce75745)

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
[... 862 lines cut ...]
set rc_expo = 70
set rc_yaw_expo = 70
set roll_rate = 75
set pitch_rate = 75
set yaw_rate = 75
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 1500
set battery_capacity_warning = 375
set battery_capacity_critical = 180
set battery_capacity_unit = MAH

#

fiam, 2018-07-12:
@lawnmower4x Please, read the help tips at the right of those values. Those angles are constrained by the maximum pitch and roll in ANGLE mode, which can be adjusted in the PID tuning tab. You have them set a 30º (the default). They're `max_angle_inclination_rll` and `max_angle_inclination_pit`.

lawnmower4x, 2018-07-12:
Thank you very much, I thought that setting was for FW. The mini is flying good now and even in angle terribly fast.

closed 2018-07-12 by lawnmower4x

fiam, 2018-07-13:
@lawnmower4x Awesome! I guess we need to make the name for those options more self-explaining.