iNavFlight/inav #3580: MATEKF405_SERVOS6: No rudder output on S6

bseishen, 2018-07-12:
Matek F405-STD with FCHUB-W
Target MATEKF405_SERVOS6 & MATEKF405
Version 1.9.1, Tried all other versions too. Only version found working correctly is a custom binary (1.8.0) from Matek's website.

Trying to get Rudder output on Pin S6 (PA8). Rudder always defaults to S5 when using Airplane mixer. No clue why. Configuration file seems ok.

`#ifdef MATEKF405_SERVOS6
    { TIM1, IO_TAG(PA8),    TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO },  //S6 DMA2_ST6 SV6`

Mixer's used: Airplane and tried to do a custom airplane mix with no avail. Just not getting PWM signals on S6.

bseishen, 2018-07-12:
@mateksys ^^

Andreas-Hellberg, 2018-07-12:
@bseishen I got rudder working on my Matek F405 STD on version 1.9.0, using S6. But I flashed the normal version not "servo 6". It also works fine on Inav 2.0.0 RC2. Version 1.9.1 did not work for me either though.

Andreas-Hellberg, 2018-07-12:
My diff on INAV 1.9.0 with working rudder on the MatekF405 build.
https://pastebin.com/hA0yEKvH

MATEKSYS, 2018-07-13:
https://github.com/iNavFlight/inav/issues/3223

bseishen, 2018-07-13:
I'll give 1.9.0 or 2.0.0 a try. Thanks everyone!

closed 2018-09-25 by stale[bot]