iNavFlight/inav #3582: Does not read the settings of the speed controllers via the Omnibusf4v3 flight controller on firmware 2.0.0 rc3.

lavaed, 2018-07-12:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

giacomo892, 2018-07-12:
Are you sure ESCs are running BLHeli?
Are you powering up the quad with battery before connecting BLHeli Suite?

Try BLHeli - Configurator from chrome app store too.

lavaed, 2018-07-12:
All you described I do. But on firmware 1.9.1 omnibusf4v3 everything works, and when I flash on 2.0.0rc3 nothing works.

digitalentity, 2018-07-26:
@lavaed what ESCs are you using? I'm going to try and reproduce the issue.

digitalentity, 2018-07-29:
@lavaed please update

digitalentity, 2018-08-05:
Can't reproduce. Tested BLHeli_S & BLHeli32 ESCs. Closing.

closed 2018-08-05 by digitalentity