iNavFlight/inav #3584: New feature request - Live current sensor calibration via OSD

bparini, 2018-07-13:
new feature request.
Hi INAV team;
Any possibility to implement calibration of the current sensor via the OSD menu?. Similar to eagle tree vector.
For those of us who do long range flights it is very critical to have the sensor calibrated.

TIA!
[image]

teckel12, 2018-07-13:
@bparini I must be missing something here.  Why would you calibrate the current sensor via OSD?  You're not talking about while flying are you?  Don't you need a reference current sensor to do the calibration?

I do current sensor calibration on the bench, with a multimeter monitoring current as I spin up the motors (with the quad strapped to a heavy object).  I do the calibration in the configurator.  Even on the bench, I wouldn't want to use OSD as I'd have to keep taking the goggles off to read the multimeter current.

Like I said, maybe I'm missing something.  Maybe I'm doing it all wrong and there's a better way?

bparini, 2018-07-23:
@teckel12 Currently I do it in the same way but I use a display monitor in the bench to make adjustments instead of goggles. With Vector OSD and with motor running, I compare the values between the OSD and the ampmeter clamp. Then I adjust the difference (all this without connecting the FC to the computer). With INAV there are differences in readings in Voltage and Current when you connect the FC to the computer. For long range flights, accuracy is required. Currently get to it with inav you could take many hours in the bench trying to reach acceptable values.

closed 2018-10-05 by stale[bot]