iNavFlight/inav #3586: iNav 2.0.0 RC3 > GPS Coords AWOL & No Rudder input for Cruise mode

moggiex, 2018-07-13:
Howdy,

I flew RC3 in a wing today and while the flight timer worked this time, the GPS coords in the OSD were missing and also I noticed that in cruise mode, the rudder stick had no output on the models direction.

**GPS**
- Yes the fonts were reuploaded after flashing the new version (everything else worked OK in the OSD)
- Yes the model did RTH fine, more than once
- All it showed was the LAT and LON icons, but no coords in the OSD

** Cruise Mode**
Today I combined this with NAV ALTHOLD as suggested in the release notes, however unlike yesterday with RC2, when using cruise mode by itself, I could use the rudder stick to change direction. So either with the combination of NAV ALTHOLD or a change in RC2 to RC3, this was lost.

I'm not sure if that was intentional or not, however I liked the the use of the rudder stick in the wing!

And no sorry no DVR recording, I just got home and found that it was corrupted :( 

I can dump my settings if desired,

Matt

fiam, 2018-07-13:
@moggiex That's because you upgraded from RC2 without doing a diff/erase/load. Please, take a `diff` or a `dump` from the CLI, then use `defaults` to delete all settings and load them again.

The reason you're not seeing the coordinates is because their length is now configurable and they have an incorrect value (we don't keep setting compatibility from RC releases, it should work fine when upgrading from 1.9.1).

Someone else will comment on the cruise bug. You should be able to change direction with the YAW stick, it hasn't been intentionally disabled.

moggiex, 2018-07-13:
@fiam re that process, yes most likely! I'll confirm in the morning.

r3df1v3, 2018-07-13:
Same thing here, upgraded to RC3, and coords are missing, time is off by one hour, and RTH is overshooting the path to go home, and the plane comes back making S turns...
Flight timer works fine.
[image]

r3df1v3, 2018-07-13:
OK, Thanks.
I fixed the GPS cords by setting osd_coordinate_digits = 8, and the time zone now should be fine tz_offset = 60 (GMT+1 for Spain)

fiam, 2018-07-13:
@r3df1v3 If you have a blackbox log, just open an issue. That way we can see what the plane was doing during RTH.

shellixyz, 2018-07-14:
@moggiex Are you sure your yaw stick is mapped and mapped to the right channel ? I can only see this reason for it not to work.

shellixyz, 2018-07-14:
@moggiex  Another reason might be that you have a bad value for `nav_fw_cruise_yaw_rate` since you didn't properly reset the configuration. The default is 20.

moggiex, 2018-07-14:
Good morning gents,

Below is a dump from the board, no changes have been made yet (ie resetting the board as suggested above).

GPS Coords, yes that looks like a setting issue as mine is set to 1, hence.... when no coords on the screen right? 

`set osd_coordinate_digits = 1`

Fixed on that one eh? YAY!

@shellixyz I just checked the value of nav_fw_cruise_yaw_rate  and thats not it, it's 20. The only thing different from yesterday is the version (RC3 from RC2) and I added in NAV ALTHOLD to the NAV CRUISE flight mode.

So I would guess, when that condition is met, rudder input is being ignored?

Settings diff below,

Matt
```
# get nav_fw_cruise_yaw_rate
nav_fw_cruise_yaw_rate = 20
Allowed range: 0 - 60
```

Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/MATEKF722 2.0.0 Jul 12 2018 / 11:32:01 (81ce75745)

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 4 0 -50 0
smix 1 4 1 -50 0
smix 2 3 0 -50 0
smix 3 3 1 50 0

# servo

# feature
feature -TELEMETRY
feature MOTOR_STOP
feature GPS
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 1 64 115200 115200 0 115200
serial 2 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1775 2100
aux 1 2 1 1300 1700
aux 2 3 5 1750 2100
aux 3 8 2 1750 2100
aux 4 19 4 1750 2100
aux 5 35 5 1750 2100
aux 6 10 1 900 1275
aux 7 27 3 1325 1725
aux 8 28 3 1700 2100
aux 9 32 4 1750 2100

# adjrange

# rxrange

# osd_layout
osd_layout 0 0 1 1 V
osd_layout 0 1 23 1 V
osd_layout 0 7 1 4 V
osd_layout 0 9 23 3 H
osd_layout 0 11 23 2 V
osd_layout 0 12 23 3 V
osd_layout 0 13 2 7 V
osd_layout 0 14 1 2 V
osd_layout 0 15 23 7 V
osd_layout 0 20 18 14 V
osd_layout 0 21 1 14 V
osd_layout 0 22 11 1 V
osd_layout 0 23 13 1 V
osd_layout 0 28 1 3 V
osd_layout 0 33 23 4 V
osd_layout 0 34 10 12 V
osd_layout 0 40 23 11 V
osd_layout 0 48 1 3 H
osd_layout 1 0 1 1 V
osd_layout 1 1 23 1 V
osd_layout 1 3 8 6 V
osd_layout 1 7 1 4 V
osd_layout 1 11 23 2 V
osd_layout 1 12 23 3 V
osd_layout 1 13 1 7 V
osd_layout 1 14 1 2 V
osd_layout 1 15 24 7 V
osd_layout 1 20 18 12 V
osd_layout 1 21 1 12 V
osd_layout 1 22 11 1 V
osd_layout 1 23 13 1 V
osd_layout 1 28 1 3 V
osd_layout 1 30 1 13 V
osd_layout 1 33 23 4 V
osd_layout 1 34 10 11 V
osd_layout 1 38 24 2 H
osd_layout 1 45 0 0 V

# master
set looptime = 2000
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 40
set acc_hardware = MPU6500
set acczero_x = -28
set acczero_y = -13
set acczero_z = 14
set accgain_x = 4090
set accgain_y = 4083
set accgain_z = 4093
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 12
set failsafe_throttle_low_delay = 100
set failsafe_procedure = RTH
set align_board_pitch = 20
set vbat_scale = 1093
set current_meter_scale = 165
set platform_type = AIRPLANE
set model_preview_type = 8
set small_angle = 180
set switch_disarm_delay = 150
set gps_sbas_mode = AUTO
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 7500
set nav_fw_bank_angle = 30
set nav_fw_climb_angle = 30
set nav_fw_loiter_radius = 6000
set osd_units = UK
set osd_stats_energy_unit = VALUE 2 OUT OF RANGE FOR osd_stats_energy_unit

set osd_time_alarm = 20
set osd_alt_alarm = 500
set osd_dist_alarm = 5000
set osd_crosshairs_style = AIRCRAFT
set osd_left_sidebar_scroll = GROUND_SPEED
set osd_right_sidebar_scroll = ALTITUDE
set osd_coordinate_digits = 1
set osd_estimations_wind_compensation = OFF

# profile
profile 1

set fw_p_pitch = 6
set fw_i_pitch = 9
set fw_ff_pitch = 52
set fw_p_roll = 6
set fw_i_roll = 6
set fw_ff_roll = 49
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set tpa_rate = 20
set tpa_breakpoint = 1600
set roll_rate = 28
set pitch_rate = 14

# battery_profile
battery_profile 1

set battery_capacity = 3000
set battery_capacity_warning = 2100
set battery_capacity_critical = 2700

moggiex, 2018-07-14:
Oh and to chip in 2p of feedback on the NAV CRUISE mode, I would "expected it" to have managed the height by default.

And yes I am assuming the cruise mode was from the original addition request I put in a while back, which mentioned that the cruise mode in the eagle tree vector (2DH), maintains direction and height at the same time.

Matt

shellixyz, 2018-07-14:
@moggiex You can use `CRUISE` with or without `ALTHOLD`. Without `ALTHOLD` you keep control of the pitch and throttle. If `nav_fw_cruise_yaw_rate = 20` and your yaw stick is mapped to the right channel it should just work (combined with `ALTHOLD` or not). Cruise mode has been tested extensively. Please make sure that when you move the yaw stick you can see the yaw channel moving in the receiver tab. You could add the cruise heading adjustment OSD element to see what effect your yaw input has. When `CRUISE` mode is enabled and you input a yaw command it should appear and display the heading change you asked and will disappear 4 seconds after you release the yaw stick.

moggiex, 2018-07-14:
Howdy,

The rudder stick is mapped correctly. 

As mentioned it worked 2 days ago with RC2, however yesterday with RC3 with NAV ALT HOLD it did not.

Not sure what else you would like me to add. It didn't work.

Matt

moggiex, 2018-07-14:
I can, but it won't be until next Friday at the earliest.

Matt

fiam, 2018-07-14:
@moggiex Do you happen to have a blackbox log of that flight? That would work too.

moggiex, 2018-07-15:
@fiam sadly no :( 

I forgot to fit a card. Bad Matt! One is fitted now and I won't forget in future, honest.

Matt

digitalentity, 2018-07-29:
@fiam @shellixyz @moggiex any update on this?

shellixyz, 2018-07-29:
@digitalentity 

I tested the coordinates are displayed correctly in dev head.

Apparently the yaw control in cruise is working I heard @moggiex mention it in https://www.youtube.com/watch?v=yPq9TwKTUZk. The only things that can prevent it to work are a bad value for `nav_fw_cruise_yaw_rate` and if the roll channel is outside of the dead band when the stick is centered since roll has priority over yaw.

digitalentity, 2018-07-30:
@shellixyz I guess we can close this issue then

digitalentity, 2018-08-15:
Closing as inactive.

closed 2018-08-15 by digitalentity