iNavFlight/inav #3589: Baro only failure on i2c on Omnibus, (F3).

iturner6168, 2018-07-14:
Omnibus F3
Readytoflyquads
2.00 RC3
Baro on i2c not working while i2c Mag is OK
Fully functional with 1.9.1 before flashing 2.00 RC3. It worked again as before after re-flashing 1.9.1
Would not recognize baro when baro set to "Auto". When set to correct part, (BMP085), Baro symbol appeared in red and no functionality.

shellixyz, 2018-07-14:
An external baro is not supported by default by the OMNIBUS target anymore due to size constraints  and since one is already present on the board.

You can build a custom version if you want to use an external baro. Here are the changes you need to make:
```
diff --git a/src/main/target/OMNIBUS/target.h b/src/main/target/OMNIBUS/target.h
index dec683247..e5458d7dc 100644
--- a/src/main/target/OMNIBUS/target.h
+++ b/src/main/target/OMNIBUS/target.h
@@ -51,6 +51,11 @@
 #define BMP280_SPI_BUS          BUS_SPI1
 #define BMP280_CS_PIN           PA13
 
+#define BARO_I2C_BUS             BUS_I2C1
+#define USE_BARO_BMP085 // External
+#define USE_BARO_BMP180 // External
+#define USE_BARO_MS5611 // External
+
 #define USE_MAG
 #define MAG_I2C_BUS             BUS_I2C1
 #define USE_MAG_HMC5883
```

I built it for you this time: 
[inav_2.0.0_RC3_OMNIBUS_ext_i2c_baro.zip](https://github.com/iNavFlight/inav/files/2194750/inav_2.0.0_RC3_OMNIBUS_ext_i2c_baro.zip)

iturner6168, 2018-07-14:
Many thanks for the response. The Omnibus is available without mag or baro to satisfy the acro and racer folks. I actually purchased the board in error and was able to use it by grafting a baro board into the case of a GPS unit which makes a really neat setup. On the subject of coding, I used to write hobby code back in the days of the Apple II (1980), when I wrote a commercial program called Speedfile entirely in Motorola 6502 machine code, including customized direct control floppy disk code and a screen editor. It took me over a year and over a thousand hours of fanatical evenings but it sold thousands. I went on to get promoted into management at my job to the point where I became completely hands off and eventually became VP of Engineering of a computer peripheral company and finally a CEO, (and therefore useless).  I am now retired and have so far avoided getting back into coding since I am so totally out of date with tools etc. My total modern experience is that I have occasionally poked some configuration changes into some Arduino code to be able to flash an ESC.  This little issue has pushed me into looking into get involved again since I really appreciate the work of the inav group! Can you point me towards a start point bearing in mind I know absolutely nothing of modern software environments but I am really, really, smart! PS. I even have a patent on an optical disk drive servo method so I have a basic understanding of servos. Thanks again for the response.

closed 2018-08-19 by DzikuVx