iNavFlight/inav #3591: mid-flight disarm - log file included

Hyperious, 2018-07-14:
Hey all, had a random disarm on a flight earlier tonight. I was out at around 3km testing the range on a new triple-feed patch I got when the thing completely disarmed on me. Thankfully I was able to glide it down in disarm mode to a small field near an intersection I knew, but nevertheless it was disnerving that this is occurring, especially since I'm in the middle of building a dedicated long range wing using inav as the base.

Attached is the log file, and the current config on the plane. The log is saying it was a stick disarm, which I know to be bullshit since there's no way I could have stick disarmed it in flight.

[LOG00054.TXT](https://github.com/iNavFlight/inav/files/2194659/LOG00054.TXT)
[cli dump.txt](https://github.com/iNavFlight/inav/files/2194660/cli.dump.txt)

giacomo892, 2018-07-14:
Hello, you didn't followed the issue template so i'm just guessing. 
You talked about "long rong" so I assume you are running R9 system.
Upgrade to 2.0RC3 and all issues with midair disarms should be solved.

Hyperious, 2018-07-14:
1.9.1, yeah, and a Q-X7. it's the same firmware as the R9. is there a conflict between the two?

giacomo892, 2018-07-15:
Are you using or not R9 Long Range system?

Hyperious, 2018-07-15:
Not the long range system, no. The RX is an xm+ 16 channel. I've updated the FC to 2.0.0RC4 since this issue as a suggestion from a few on Reddit that said that 1.9.1 had some bugs with XM+ rx units and random disarms. Hopefully the new release solves the problem.