iNavFlight/inav #3604: EZ GUI not displaying active modes in 2.0.0 RC3

flyfisher2, 2018-07-15:
MATEK F405-WING, MATEKF405SE target
QUAD X
Release 2.0.0 RC3 (probably all 2.0.0 release candidates)

In the DASHBOARD 1 page of EZ GUI, no modes are displayed in the Active modes panel.
Also if the MODES tab is pressed, the application crashes.
All other functions seem to still work OK.

giacomo892, 2018-07-15:
EZGUI is no more developed. There is nothing we can do about it.

flyfisher2, 2018-07-15:
I know, but it still works with 1.9.
Are you saying that iNav has been changed in 2.0 to be incompatible with EZGUI?

stronnag, 2018-07-16:
There are publicly documented changes to iNav (see the release notes). We do not have the ezgui source code so we cannot tell how they might affect a closed source application. 

On the other hand, the author of ezgui has open access to the iNav source code. This ticket belongs in the ezgui issue tracker, not here.

flyfisher2, 2018-07-16:
Unfortunately the author seems to have taken the money and run away from this application.
Back to lugging a notebook to the flying field.
Thanks anyway.

closed 2018-07-16 by flyfisher2