iNavFlight/inav #3607: minimum RSSI value is not displayed correctly

jaames74, 2018-07-15:
Fresh flashed Inav2.0RC4 on spracingF3, frsky X8R with pwm-to-analog-converter.
X8R rssi is 0-3.3V tested with multimeter. Shows as 32% to 100% in Inav configurator.
Cannot be configured properly with RSSI_SCALE.

Also tested with inav 1.9.1 - same behavior.

teckel12, 2018-07-15:
Have you tried sending the RSSI back from your transmitter to your FC? I could never get RSSI to work correct directly from receiver to FC.

jaames74, 2018-07-16:
@teckel12 i don't understand what you are talking about. My radio is very old - Spektrum DX7 with Frsky DHT module. Analog RSSI works OK with other firmwares such as ardupilot or betaflight. Why not with INAV??

r3df1v3, 2018-07-16:
Same here with a Thomas Scherrer LRS receiver.
If I set the RRSI to read max value at 99%, then when switching off the TX, never goes down more than 75%.
Please, fix that  scale!!

closed 2018-07-19 by jaames74

fiam, 2018-07-19:
Fixed in #3613

commit c34a408f (2018-07-16): Add rssi_offset, apply offset and scale to all protocols

This allows any possible received RSSI to be mapped to [0, 100],
no matter the range and offset of value emitted by the RX.

Also, apply the RSSI scale and offset to all RSSI sources except
via MSP (via MSP we expect the sender to be well behaved and send
only values in [0, RSSI_MAX_VALUE]).

Fixes #2898
Fixes #3497
Fixes #3607