iNavFlight/inav #3610: Spektrum Sat receiver not working on 2.0.0(rc1 - rc4)

smaller09, 2018-07-16:
Omnibus F4 PRO V2
spektrum sat 
Fully functional with 1.9.1 before flashing 2.00 RC3. Would not show any signal in Receiver tab, also try 2.0.0-rc1,rc2 and rc4.
Worked again after reflashing 1.9.1

stronnag, 2018-07-16:
This is target specific problem, as I flew spek sat on RC4 on OmnibusF3 and MatekF405 today.

fiam, 2018-07-16:
I don't think anything affecting Spektrum receivers was [intentionally] changed in 2.0.0

stronnag, 2018-07-16:
It wasn't, they work perfectly (orl as perfectly as Spek Sat can) here on 5 different iNav machines, all of which have been extensively tested during 2.0 RCs.

digitalentity, 2018-07-16:
@smaller09 did you follow the 2.0 update gudilines (specifically "flash with full erase and reconfigure" recommendation)?

smaller09, 2018-07-17:
@digitalentity I did full erase everytime,  and flashed 2.0 rc4 again, but nothing change ,still no signal in receiver tab. 
The configuation, only set mixer and rx.
# diff

# version
# INAV/OMNIBUSF4PRO 2.0.0 Jul 14 2018 / 22:40:31 (f2cf315ad)

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo

# feature

# beeper

# map

# serial
serial 0 64 115200 38400 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# osd_layout

# master
set acc_hardware = MPU6000
set acczero_x = 76
set acczero_y = 18
set acczero_z = -195
set accgain_x = 4071
set accgain_y = 4082
set accgain_z = 4049
set mag_hardware = HMC5883
set magzero_x = 136
set magzero_y = -190
set magzero_z = 483
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SPEK2048
set failsafe_procedure = RTH
set platform_type = AIRPLANE
set model_preview_type = 8

# profile
profile 1

# battery_profile
battery_profile 1

smaller09, 2018-07-17:
I tried rebind the sat in dsm2/1024, dsmx/1024, dsm2/2048, dsmx/2048, still no signal. 
Also, there's a small bug in OMNIBUSF4  target. Spektrum sat port is uart1, but the bind pin was set to uart3 rx in target.h, I have to recompile the firmware to make sat into bind mode.

#define BIND_PIN                PA10 // USART1 RX   **(Orignal was PB11 // USART3 RX)**

smaller09, 2018-07-18:
I flashed another FC (spracingf3) to 2.0.0 to test the sat receiver, and it works, so  it should be a target specific problem.

smaller09, 2018-07-18:
@digitalentity  I reverted the code to find which commit cause that problem, and found that was caused by commit 9e6e2b3f6... Timer driver refactoring    #3120 , but I don't have the ability to find out the reason.

digitalentity, 2018-07-29:
SBUS verified to work on UART1 on two different OmnibusF4 Pro boards both with 1.9.1 and 2.0

digitalentity, 2018-07-29:
@smaller09 can you post a photo of a board where SBUS input doesn't work? Did you remove PPM jumper as instructed in board manual?

smaller09, 2018-07-30:
[https://www.dronetrest.com/uploads/db5290/optimized/2X/3/39f8975bf77b2ab1d24514b46f72255097da3206_1_690x387.jpg](url)

[https://sep.yimg.com/ty/cdn/yhst-90268561309754/NA0065-01.jpg](url)

I‘m using the left one omnibus f4 pro v2.
this fc does not have the PPM/SBUS jumper onboard,  just two res connect to J7（R19,R20), one is wired to PB8 (ppm input, btw: ppm works on 2.0) another is wired to inventer and then to PA10.  I  try connect the receiver to PPM/SBUS header, and mod my receiver to get a un-invent sbus signal wired to J5 (uart1), both works on 1.9.1 and does not on 2.0
I will check the status of PB8 in fw2.0, if the pin set to HIGH or LOW, then it will effect the rx input

smaller09, 2018-07-30:
Ok， I removed R20 and the RX works on 2.0, that will be the problem. but I‘m confused that the original setup works on 1.9.1

closed 2018-07-30 by smaller09

digitalentity, 2018-07-30:
Ok, this might be related to changed timer default pin init. In such case https://github.com/iNavFlight/inav/pull/3673 might help.

sq9fk, 2018-09-07:
From Inav 2.0.0 was changed UART definition in target.c
Please look at my comment https://github.com/iNavFlight/inav/issues/3574#issuecomment-419371066

Focused on:
inav/src/main/target/OMNIBUSF4/target.h

```
//Commneted out because OMNIBUSF4PRO V2 doesn't have pin for inverter.
//#define USE_UART_INVERTER
```

and

```
#if !defined(OMNIBUSF4V3)
//Commneted out because OMNIBUSF4PRO V2 doesn't have pin for inverter.
//#define INVERTER_PIN_UART1_RX PC0 // PC0 has never been used as inverter control on genuine OMNIBUS F4 variants, but leave it as is since some clones actually implement it.
#endif
```