iNavFlight/inav #3627: Possible bug: no GPS coords sent to DJI Occusync OSD on DJI Goggles RE

VoicOfReason, 2018-07-19:
I realize this may be a problem on DJI's side of things, but in the DJI Occusync (air module with goggles) system, there is an OSD that shows direction home, # of sats, speed, alt, lat/long. DJI pulls this data from the FC and puts in their OSD in the DJI Goggles Racing Edition. All this data works except for lat/long which display as 0's. Just thought I'd mention this even though it may be DJI's fault.

My FC: SP Racing F4 EVO
GPS: BN-880
Occusync Air Module
DJI Goggles RE
Build: iNav 2.0 RC4

giacomo892, 2018-07-25:
I've red DJI ocusync uses MSP for telemetry, so if no coordinates are displayed is DJI fault not asking/decoding them correctly.

digitalentity, 2018-07-25:
I've contacted DJI. Let's see if they could provide some insight on this matter.

Redshifft, 2018-07-25:
Be interesting to see if you actually get a reply, there has to be a first time for everything :)

digitalentity, 2019-01-06:
Closing, as DJI is unwilling to cooperate.

closed 2019-01-06 by digitalentity