iNavFlight/inav #3628: async gyro, unable to set anything beside 8K samp frequency

giacomo892, 2018-07-19:
Quick note.

It is not possibile, when choosing to run ASYNC (ALL/GYRO), to select a sample frequency that is not 8K for the gyro.

INAV 2.0.0RC4 both F3/F4

DzikuVx, 2018-07-20:
Yes, it changed and we should remove this setting. FC picks "the best" sampling frequency

giacomo892, 2018-07-20:
ack that, but basically it means it will run at 8K and severly restrict the looptime on F3 targets.

digitalentity, 2018-07-20:
Please share dump of your settings

giacomo892, 2018-07-20:
# version
# INAV/MATEKF405SE 2.0.0 Jun 29 2018 / 12:31:28 (a8415e89c)

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 0 1000 2000 1500 100 
servo 1 1000 2000 1500 100 
servo 2 1000 2000 1500 100 
servo 3 1000 2000 1500 100 
servo 4 1000 2000 1500 100 
servo 5 1000 2000 1500 100 
servo 6 1000 2000 1500 100 
servo 7 1000 2000 1500 100 

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
[... 846 lines cut ...]
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

I've select async, 4K gyro and 2K loop in the configurator.

giacomo892, 2018-07-20:
[image]
configurator ends up showing this

digitalentity, 2018-07-20:
Thanks @giacomo892. What's shown at CLI `tasks` ?
This might be a bug.

giacomo892, 2018-07-23:
[image]

giacomo892, 2018-07-23:
Seems like GYRO task when async is selected run at PID loop frequency no matter what.

digitalentity, 2018-07-26:
Ok, this was broken a while ago when we deprecated `gyro_sync_denom`. There is no such setting anymore but Configurator is still considering it when changing async mode. Similarly, when making decision about gyro sampling frequency firmware only looks at `looptime`.

Suggestion - introduce new setting to explicitly set desired gyro frequency and modify firmware and configurator accordingly.

digitalentity, 2018-07-26:
This is too big of a change, postponing to 2.1

digitalentity, 2018-07-26:
Yes, it's better to run synced 4K

giacomo892, 2018-07-26:
Unsure if we should suggested not to run async in 2.0 👍

DzikuVx, 2019-01-05:
Async modes are gone in 2.1, so closing

closed 2019-01-05 by DzikuVx