iNavFlight/inav #3629: BLHeli_32 ESC telemetry

xermalk, 2018-07-19:
Reopening this feature request. https://github.com/iNavFlight/inav/issues/2710

ESC telemetry (CURR/RPM/TEMP) would be a **_massive_** benefit for wing wiring and and allow for extra warnings if esc is to hot or the motor rpm differs to match from the throttle signal.

Its also one more reason to use the new BLHELI32 60-70-80A escs in planes.  Some of them are as light as 4.5g ! vs a regular plane esc are usualy 60-80g for similar specs.

digitalentity, 2018-07-20:
Duplicate of #2710. Reopened.

closed 2018-07-20 by digitalentity

putimir, 2019-08-14:
Hey @digitalentity, what is this: 
[image]