iNavFlight/inav #3637: 2.0.0 RC4 - Omnibus F4Sd/Pro - FC does not see CrossFire

romanas, 2018-07-21:
Tested with all pre versions, RC3 and RC4, FC can't see Crossfire inputs. Receiver tab shows all at 1500.
Works fine with 1.9.1 iNav.

Tested with clean installs/presets/custom builds RC3.

wx4cb, 2018-07-22:
have you swapped the TX/RX wires? are you using CSRF? if so have you setup the receiver correctly and also told the FC that you're using CSRF?

i've just put 2.0.0/RC4 on a brand new F405-Wing and it worked first time

CrashTestDummy-CTD, 2018-07-22:
Got the same problem...with omnibus fc

leeph, 2018-07-23:
I have the same issue with a PPM RX. Works fine in 1.9.1 on Matek F722 - flash to 2.0.0 and apply the exact same port & receiver config and there just is no input from the receiver under 2.0.0, RC1, 2, 3 and 4 all tested and no worky.

digitalentity, 2018-07-25:
@romanas please post complete dump of non-working RC4.

digitalentity, 2018-07-26:
Tested on OMNIBUSF4 board, Crossfire on UART6. No issues, CRSF working perfectly.

romanas, 2018-07-26:
While you are there, could you test same setup with 1.9.1 ?

digitalentity, 2018-07-26:
Already did. Working as well.

digitalentity, 2018-07-29:
Regarding CRSF/SBUS on OmnibusF4 PRO boards. Issue confirmed and existing on ALL versions of INAV. It appears that PPM/SBUS input on that board is always inverted regardless of what firmware is asking.

blurcak, 2018-08-01:
Hi i have simillar problem iBus+Uart1 (OmnibusF4pro)
Yesterday i made upgrade from 1.9.1 to 2.0.0RC4 and receiver is not working
Reflash to 1.9.1 (inav_1.9.1_OMNIBUSF4PRO.hex) Fullchip erase  RX is ok 
But on 2.0.0RC4 (inav_2.0.0-rc4_OMNIBUSF4PRO.hex) Fullchip erase no movement in receiver tab.
I have OmnibusF4-Pro-0A board RX connected to J5(Uart1) connector.

digitalentity, 2018-08-01:
@blurcak can you test this hex?

[inav_2.0.0_OMNIBUSF4PRO.hex.zip](https://github.com/iNavFlight/inav/files/2250374/inav_2.0.0_OMNIBUSF4PRO.hex.zip)

blurcak, 2018-08-01:
Sorry for delay, its working 👍

digitalentity, 2018-08-01:
@blurcak roger that. Merging the fix!

closed 2018-08-01 by digitalentity

romanas, 2018-08-01:
Ok, thanks will try to test your custom hex also.

leeph, 2018-08-01:
Is this fix in any way applicable to my Matek F722 board and PPM? Because the symptom is exactly the same as what's being described above for Crossfire etc on Omnibus boards...

waveydavey420, 2018-10-28:
Same here. Crsf is working on 1.9.1, but when I switch to 2.0 no Rx. Also on 1.9.1 s3 output sevo is fluttering, does not work.  Using f405 wing fc

teckel12, 2018-10-30:
@waveydavey420 I believe you'd need a pre-release flash for the F405 to verify it works or not. You could compile your own for your target (see the cross-linked PR), or wait for the next release.