iNavFlight/inav #3638: Impossible to arm after very long flight

jreis1, 2018-07-21:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

digitalentity, 2018-07-22:
@jreis1 a dump of configuration and blackbox log (if available) would help.

jreis1, 2018-07-22:
[dump.txt](https://github.com/iNavFlight/inav/files/2217254/dump.txt)

dump file attached. Tomorrow will add a DVR of the last minutes.

jreis1, 2018-07-23:
@digitalentity  please find dump file attached and DVR here: https://youtu.be/6Eb78SvS9e0?t=20s

No blackbox logo. 

The DVR ends with a "rc rx link lost" since i've turned off my taranis.

marbalon, 2018-08-05:
I was on holiday and also have the same problem, so after come back I reproduced problem on my desk. Ignore voltage etc because the model is powered using USB.

Scenario.
Arm - wait > 60-70min, enable blackbox on switch and disarm. In logs FC shows that flag ARM disappear, but in GUI I still see icon motor armed.

Diff and blackbox attached.

[drewniak-2.0.txt](https://github.com/iNavFlight/inav/files/2260687/drewniak-2.0.txt)
[unable to dissarm.TXT](https://github.com/iNavFlight/inav/files/2260688/unable.to.dissarm.TXT)

shellixyz, 2018-08-05:
What version of the configurator are you using ? It is only doing that after you leave it armed for a long time ? Can you spin the motor after you try to disarm ?

marbalon, 2018-08-06:
GUI version is 2.0.0-rc3. I will let you know about other  question after another test but I've just upload RC5.

On RC4 on my holiday I had danger situation because after long fly I landed away from me and model got FS - normal, start to spin motor on the ground but this is ok with my configuration. But the problem was when I receive the link (OSD shows 99%) I can't get control back. No response to any sticks/switches - I need to unplug battery to turn of the motor.

marbalon, 2018-08-06:
Ok, I've just made another test on my desk. Version RC5, problem seems to be realy danger because if you get failsafe after 60min you will can't control you plane anymore.

Scenario for today:
1. FC connected via USB without battery, armed using switch, and throttle raised up.
2. Turn off battery and wait 60-70min. 
3. Turn on transmitter - GUI still shows failsafe icon but I can see on receiver tab that all channels is working, but I can't disarm model, no response to sticks etc. FC control servos and motor.
4. what is strange I can enable blackbox on switch to start logging.
5. Move arm stick to disarm model and throttle down but without effect.
6. Turn off bb logging using switch and download logs.

Logs attached. Configuration the same like above.
[unable2.TXT](https://github.com/iNavFlight/inav/files/2262056/unable2.TXT)

marbalon, 2018-08-06:
And test nr3. Same scenario as above but I've added kill switch on the same channel as arm, so if I try to disarm I enable killswitch (with some slow down/up in taranis).

Now I can disarm model but can't arm it again because failsafe icon in GUI is enabled. All channels are OK, RSSI is 100%, but can't do anything.
[unable3.TXT](https://github.com/iNavFlight/inav/files/2262925/unable3.TXT)

jreis1, 2018-08-06:
nice work @marbalon . thanks for your time. I'm glad not to be the only one with this issue.

digitalentity, 2018-08-06:
Fix ready. @marbalon could you please test #3703

digitalentity, 2018-08-06:
Fix merged.

PR #3703 (2018-08-06): Fix RX timekeeping overflow
Fix #3638
closed 2018-08-06 by digitalentity