iNavFlight/inav #3640: BUG in Mission Planner on android (latest version)

wx4cb, 2018-07-22:
I've not found a mission planner github or anything like that to report this to so my apologies in advance if i'm posting in the wrong spot.

I seem to have found a bug in either the mavlink implementation or the misison planner app at least on android.

Hardware: 

Matek F405 Wing - Inav v 2.0.0 RC4
TBS Crossfire nano RX, and Full size TX
Channel map is CSRF TX/RX (CH 1+2) and Mavlink TX/RX ( CH 3+4)
Mavlink telemetry enabled on FC

Issue:

After calibrating everything accelerometer-wise, I connected via bluetooth from my galaxy S10 tablet running the latest android release for it (not sure what number).

the "3D model" in the inav configurator (also the latest) operates correct as regards to pitch/roll etc.

The mission planner artificial horizon however is reporting the opposite as regards to pitch. when it's pitched up it's pointing down and vice-versa. Roll is ok and correct however.

I have uploaded a short video to youtube to show this issue: https://youtu.be/SSesDzjhJk4

shellixyz, 2018-07-22:
The Mission Planner app for Android is closed source and has not been written by iNav developers. You should contact the author about this issue: contact@ez-gui.com

wx4cb, 2018-07-22:
thanks shelly, appreciated

closed 2018-07-22 by wx4cb