iNavFlight/inav #3642: 2.0 RC4 Arming issue after CMS save reboot

lo4wai, 2018-07-22:
After save & reboot from CMS, motors do not spin after arming.  However stat will still appear during disarm.

In 1.9.x, CMS save & reboot produce ESC last startup tone, but in 2.0 RC4, they are gone.

giacomo892, 2018-07-22:
Are you running BlHeli 32 ESCs?

lo4wai, 2018-07-22:
Yes, I am running DYS Aria with Matek F405 CTR

digitalentity, 2018-07-22:
I've seen this issue as well. It seems ESCs get into some weird state when FC is soft-rebooted.

lo4wai, 2018-07-23:
Reboot in CLI and configurator (MSP I guess) are fine, motors will spin when armed.
I found little difference in code that CLI and MSP delay for 1000ms but CMS only for 200ms

I try to check out RC4 tag as a branch, change it to 1000 and build using docker, but fail with

>$./build.sh MATEKF405
Building target MATEKF405
settings.yaml -> settings_generated.h, settings_generated.c 
%% startup_stm32f40xx.s 
make: *** No rule to make target 'lib/main/CMSIS/CM4/DeviceSupport/ST/STM32F4xx/stm32f4xx_gpio.c', needed by 'obj/main/MATEKF405/stm32f4xx_gpio.o'.  Stop.

I can build the master branch, but seems it is not 2.0RC4

giacomo892, 2018-07-23:
Using PWM instead of ONESHOT/MULTISHOT makes this work... but PWM so any other things doesn't work well! Still an interesting fact

digitalentity, 2018-07-23:
Fixed by #3645, closing.

PR #3645 (2018-07-24): Unify reboot sequence across all use-cases
Fixes #3642
closed 2018-07-23 by digitalentity