iNavFlight/inav #3643: No GPS Sats suddenly with 2.0 RC4

VoicOfReason, 2018-07-23:
I was Getting up to 18 sats and flying well but after RC4, the GPS coordinates DO display and cycle with different numbers as if it’s trying to get a sat fix, but it never even gets 1 satellite fix. No crashes. I even Swapped all the wiring and swapped the GPS for a new, identical GPS. I’m running RC4 with SP Racing F4 EVO. Was flying perfect.

marbalon, 2018-07-23:
I have the same issue when on RC4. 9satelites , then 0 and again 9. So
looks like communication problem.

stronnag, 2018-07-23:
I've seen this too. In the middle of swapping some hardware around; but I doubt that's the cause. MATEKF405.

VoicOfReason, 2018-07-23:
So maybe we can elevate this issue and get it looked at and fixed soon? I’m on a trip to the mountains and haven’t been able to fly at all because of this (my fault partly for using beta firmware).

VoicOfReason, 2018-07-23:
I tried RC4 on a Mateo F722 also and got the same exact issue.

stronnag, 2018-07-23:
I'm in two (or more) minds on this one; it could just be the GPS having a 'bad hair day'
[image]

digitalentity, 2018-07-23:
I'm attempting to reproduce

VoicOfReason, 2018-07-23:
I tried 2 GPS’s and it 2 different parts of the country since I’m on a
trip. First in Arkansas and then in Montana. Also tried different times of
day.

VoicOfReason, 2018-07-23:
When I get A chance to downgrade to RC3 I’ll also see if the problem
resolves.

W. Peter Freund
Adjunct Instructor - Philosophy
Northwest Arkansas Community College
----
“Hold no knowledge in contempt, blush to learn from no man, once you have
attained knowledge, look down on no-one.” (St. Victor)

stronnag, 2018-07-23:
And in the furtherance  of statistics that prove nothing, here's my mwp logs for numsat = 0 (it's orders of magnitude easier to grep JSON than a binary log). There are definitely more instances later in the year; I'm not sure that is entirely weather / daylight related as I fly 3  - 5 times a week most of the year. It's also not hardware related as it spans 3 different machines.
```
$ grep -lr --include \*2018\*.log 'numsat":0' . | sort
./2018-02-19_custom/mwp_2018-02-19_145504.log
./2018-03-10_tri/mwp_2018-03-10_161526.log
./2018-04-28_tri/mwp_2018-04-28_135125.log
./2018-05-13_quad/mwp_2018-05-13_135807.log
./2018-05-25_quad/mwp_2018-05-25_160803.log
./2018-06-10_custom/mwp_2018-06-10_141039.log
./2018-06-10_custom/mwp_2018-06-10_143039.log
./2018-06-11_custom/mwp_2018-06-11_162553.log
./2018-06-11_custom/mwp_2018-06-11_164129.log
./2018-06-13_custom/mwp_2018-06-13_141741.log
./2018-06-13_custom/mwp_2018-06-13_142756.log
./2018-06-13_custom/mwp_2018-06-13_162034.log
./2018-06-29_tri/mwp_2018-06-29_152051.log
./2018-07-01_tri/mwp_2018-07-01_143721.log
./2018-07-04_custom/mwp_2018-07-04_142518.log
./2018-07-08_tri/mwp_2018-07-08_145235.log
./2018-07-11_custom/mwp_2018-07-11_151909.log
./2018-07-11_custom/mwp_2018-07-11_164118.log
./2018-07-16_custom/mwp_2018-07-16_162619.log
./2018-07-22_tri/mwp_2018-07-22_143707.log
./2018-07-22_tri/mwp_2018-07-22_151631.log
```
There are 21 instances this year (all development branch) of the 597 logs for 2018 (so 3.5%). Two of instances predate development being tagged as 2.0.0.

This is probably not helpful, but good luck in reproducing.

digitalentity, 2018-07-23:
Unable to reproduce. 1 hour of rock-solid fix with 8-11 sats.

stronnag, 2018-07-23:
That doesn't surprise me, I can fly for days without issue; then bang either random sat behaviour or #3267 like behaviour (on craft that have been flying iNav stably for several years). Maybe the hardware has gone bad, but at the moment I have very little confidence in 2.0.0 (and you will appreciate how much it pains me to write that). Further hardware fiddling and downgrades to follow.

VoicOfReason, 2018-07-23:
So, stronag, you are seeing 0 sats issue also? Wasn’t sure if you confirmed
this.

VoicOfReason, 2018-07-23:
Here’s what it looks like on my taranis. Stays this way indefinitely. Like I said, tried it in several states and different days/times:

https://youtu.be/iEzS3n7tzyY

digitalentity, 2018-07-24:
@VoicOfReason don't see the described issue. 
GPS just don't have a fix yet **OR** problem is sat count reporting via telemetry.

stronnag, 2018-07-24:
The issue that I and @marbalon are reporting appears to a the case where the GPS loses the fix. 

In the analysis following, 488 blackbox files (just my 2018 records:), comprising  680 flight logs results in 35 zero-satellite reports. 
None of these outages last more than 300ms and the GPS recovers. 

The zero-sat reports are obviously from the GPS, as the fixtype (0) and HDOP (9999) are consistent with raw data from the GPS (not synthesised by iNav). 

The **evidence** demonstrates that mid-flight occurrence of "X satellites ... 0 satellites ... X satellites" is not an iNav issue, it's a transient GPS issue that iNav reports correctly.

```
$ inav_0sat.rb $(find /t/inav -ipath "*2018*TXT")
****** /t/inav/2018-06-29_tri/LOG00007.TXT index = 1
 GPS 0-sat event for 0.08s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=10.63s
 Resume: numsats=18, fixtype=2, hdop=109, time=10.71s
 GPS 0-sat event for 0.12s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=12.91s
 Resume: numsats=19, fixtype=2, hdop=105, time=13.03s
****** /t/inav/2018-07-22_custom/LOG00005.TXT index = 1
 GPS 0-sat event for 0.12s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=407.79s
 Resume: numsats=19, fixtype=2, hdop=105, time=407.91s
****** /t/inav/2018-07-01_tri/LOG00008.TXT index = 1
 GPS 0-sat event for 0.12s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=2.33s
 Resume: numsats=17, fixtype=2, hdop=130, time=2.45s
****** /t/inav/2018-05-25_quad/BBL_2018-05-25_170232.TXT index = 2
 GPS 0-sat event for 0.24s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=16.88s
 Resume: numsats=16, fixtype=2, hdop=123, time=17.12s
****** /t/inav/2018-03-10_tri/LOG00007.TXT index = 1
 GPS 0-sat event for 0.20s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=5.29s
 Resume: numsats=16, fixtype=2, hdop=136, time=5.49s
****** /t/inav/2018-06-10_custom/BBL_2018-06-10_160130.TXT index = 3
 GPS 0-sat event for 0.08s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=3.99s
 Resume: numsats=13, fixtype=2, hdop=122, time=4.07s
 GPS 0-sat event for 0.12s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=31.97s
 Resume: numsats=15, fixtype=2, hdop=109, time=32.09s
 GPS 0-sat event for 0.08s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=206.05s
 Resume: numsats=16, fixtype=2, hdop=101, time=206.13s
****** /t/inav/2018-06-10_custom/BBL_2018-06-10_160130.TXT index = 6
 GPS 0-sat event for 0.28s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=230.60s
 Resume: numsats=13, fixtype=2, hdop=125, time=230.88s
****** /t/inav/2018-02-19_custom/LOG00003.TXT index = 1
 GPS 0-sat event for 0.20s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=4.03s
 Resume: numsats=13, fixtype=2, hdop=138, time=4.23s
 GPS 0-sat event for 0.08s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=4.35s
 Resume: numsats=10, fixtype=2, hdop=193, time=4.43s
****** /t/inav/2018-07-08_tri/LOG00005.TXT index = 1
 GPS 0-sat event for 0.12s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=7.70s
 Resume: numsats=19, fixtype=2, hdop=107, time=7.82s
****** /t/inav/2018-01-26_tri/LOG00007.TXT index = 1
 GPS 0-sat event for 0.12s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=4.07s
 Resume: numsats=16, fixtype=2, hdop=109, time=4.19s
****** /t/inav/2018-07-16_custom/LOG00007.TXT index = 1
 GPS 0-sat event for 0.20s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=3.67s
 Resume: numsats=18, fixtype=2, hdop=117, time=3.87s
****** /t/inav/2018-01-28_tri/LOG00006.TXT index = 1
 GPS 0-sat event for 0.08s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=4.86s
[... 52 lines cut ...]
 GPS 0-sat event for 0.08s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=30.47s
 Resume: numsats=15, fixtype=2, hdop=107, time=30.55s
****** /t/inav/2018-06-13_custom/BBL_2018-06-13_161833.TXT index = 5
 GPS 0-sat event for 0.20s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=3.47s
 Resume: numsats=13, fixtype=2, hdop=135, time=3.67s
****** /t/inav/2018-06-13_custom/BBL_2018-06-13_161833.TXT index = 6
 GPS 0-sat event for 0.20s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=3.53s
 Resume: numsats=14, fixtype=2, hdop=122, time=3.73s
****** /t/inav/2018-07-21_custom/LOG00011.TXT index = 1
 GPS 0-sat event for 0.08s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=2.50s
 Resume: numsats=18, fixtype=2, hdop=117, time=2.58s
****** /t/inav/2018-07-11_custom/BBL_2018-07-11_170806.TXT index = 3
 GPS 0-sat event for 0.12s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=3.51s
 Resume: numsats=16, fixtype=2, hdop=126, time=3.63s
****** /t/inav/2018-07-11_custom/BBL_2018-07-11_162725.TXT index = 2
 GPS 0-sat event for 0.20s duration
 Event : numsats=0, fixtype=0, hdop=9999, time=286.11s
 Resume: numsats=17, fixtype=2, hdop=110, time=286.31s
Files 488, Flight logs 680, Zero-sat records 35
```

digitalentity, 2018-07-24:
That was my conclusion as well - most likely GPS-related issue, not INAV-related.

stronnag, 2018-07-24:
Indeed, just to make it even more blindingly obvious, here's a short rerun with the "immediately before" state as well.
```
****** /t/inav/2018-06-29_tri/LOG00007.TXT index = 1
 GPS 0-sat event of 0.08s duration
 Before : numsats=17, fixtype=2, hdop=120, time=10.63s
 Event : numsats=0, fixtype=0, hdop=9999, time=10.63s
 Resume: numsats=18, fixtype=2, hdop=109, time=10.71s
 GPS 0-sat event of 0.12s duration
 Before : numsats=19, fixtype=2, hdop=105, time=12.91s
 Event : numsats=0, fixtype=0, hdop=9999, time=12.91s
 Resume: numsats=19, fixtype=2, hdop=105, time=13.03s
****** /t/inav/2018-07-22_custom/LOG00005.TXT index = 1
 GPS 0-sat event of 0.12s duration
 Before : numsats=19, fixtype=2, hdop=105, time=407.78s
 Event : numsats=0, fixtype=0, hdop=9999, time=407.79s
 Resume: numsats=19, fixtype=2, hdop=105, time=407.91s

```

VoicOfReason, 2018-07-25:
Ok everyone, I have an update. I can absolutely confirm that RC4 is responsible for 0 sats (forever) on my quad. I tried 4-5 times on my trip out west (in different locations across the country from Arkansas to Montana) to get even 1 satellite on RC4. No matter how long it sat, it never got a sat. I even messed around with all kinds of settings in iNav. Tonight, I finally downgraded to RC1 using my kid's Chromebook. Boom! No config changes. GPS worked again. Got 4 sats within 1 minute! Everything is fixed. So, there is absolutely a serious issue going on here.

Here's my setup:
BN-880
SP Racing F4 EVO
DJI Occusync Air MSP
Another MSP port for MobileFlight data via Crossfire Micro (channels 3-4)
Crossfire Micro Tx 2.0
TBS G2 frame
32bit ESC

If there is any other information/data you need from me, let me know. No doubt, RC3 would have fixed it, as well, since I was flying stable on RC3 before my trip, but iNav on Chromebook only showed RC1 as an option to downgrade to.

VoicOfReason, 2018-07-25:
> @VoicOfReason <https://github.com/VoicOfReason> diff between RC3 and RC4
> is quite small and doesn't touch anything GPS-related:
> 2.0.0-RC3...2.0.0-RC4
>
> I strongly doubt it's an RC4-specific issue.

I’ll keep looking into the issue and update here if I find The cause.

shellixyz, 2018-07-26:
I can't reproduce either. I didn't see anything different in how the GPS is working between any of the 2.0 RCs. @VoicOfReason @marbalon Are you using a RC Split camera ? They are producing EMI that can affect the GPS reception.

VoicOfReason, 2018-07-26:
Yes I am. How can I deal With split interference? The camera itself is
right next to the GPS.

DeltaVelocity101, 2018-07-27:
I am having a very similar issue on an F405 on 1.9.1 (03a5c1922) GPS appears to hop from ~6-9 sats to zero then flip back to 6-9. It does this quite quickly and seemingly with no pattern. What GPS are you using? Perhaps this is an F405 + GPS issue?

shellixyz, 2018-07-27:
@VoicOfReason I guess you can try to shield it but I don't know how easy it would be and how much weight it would add to your aircraft. The easier is probably to move the GPS module away.

@DeltaVelocity101 I'm using M8N modules. You should try to reset your GPS settings to the factory defaults by connecting it to a computer through a USB-Serial converter and using UBlox U-Center. There is a reset button on the right of the default toolbar layout.

DeltaVelocity101, 2018-07-27:
Thanks @shellixyz, although I thought U-Centre was defunct now?

shellixyz, 2018-07-27:
I don't think it is: https://www.u-blox.com/en/product/u-center-windows

digitalentity, 2018-07-28:
Most action cameras are known of emitting crazy amounts of RF noise that cripples UHF LRS systems and GPS. The cable between camera sensor and CPU board is usually the source of interference. 
I tested a number of cameras for interference and the only camera that had no apparent effect on GPS/LRS functionality was GoPro. All the rest affect GPS, sometimes to total inability to maintain a fix.

giacomo892, 2018-07-28:
@digitalentity consider the new GoPro have even integrated GPS so they should be well tested in that band

VoicOfReason, 2018-07-28:
Update: seems my RunCam split was preventing GPS from working. So, I dont
see any reason RC4 is the problem anymore. I took Off the Split and problem
vanished. Not sure what to do about that problem.

shellixyz, 2018-07-28:
I'm using Firefly Q6 cameras and I don't get any GPS or RC link issue with them.

digitalentity, 2018-07-29:
I'm closing this issue as GPS sats problem is clearly not firmware-related but induced by Runcam Split EMI noise.

closed 2018-07-29 by digitalentity

bokie1, 2018-07-29:
Sadly I'm not using runcam split.

digitalentity, 2018-07-29:
@bokie1 please be more speific about "can't get any satellites". 

Which GPS are you using?
Which UART is it connected?
Which protocol?
Is GPS detected by INAV?
How are you observing zero satellites - Configurator/OSD/telemetry?

bokie1, 2018-07-29:
Which GPS are you using? - BN-220
Which UART is it connected? - UART3
Which protocol? - Ublox
Is GPS detected by INAV? - Yes
How are you observing zero satellites - Configurator/OSD/telemetry? - All 3.

Everything was was working in 1.9.1.

digitalentity, 2018-07-29:
@bokie1 can you provide config dump of non-working setup please.