iNavFlight/inav #3644: CPU overload on Mateo F722 at 8khz

VoicOfReason, 2018-07-23:
I just tried running 8khz/8khz on my Matek F722 build under 2.0RC4 and got such high CPU usage that the board barely functioned. I am able to run the same exact config with 8khz/8khz on an F4 board with no issues at all, on RC4.

DerEnderKeks, 2018-09-16:
I have the same problem. It says the CPU usage is ~1000+% (yes, 3 zeros).

VoicOfReason, 2018-09-16:
Anyone know anything about this issue? Not yet resolved for me.

closed 2018-11-29 by stale[bot]

Nihilistic, 2019-01-12:
Yeah, I'm having the same issue on a Matek F722-SE with iNAV 2.1 RC1. I'm thinking I may need to make a youtube video and warn the entire universe not to buy this piece of shit board.

teckel12, 2019-01-12:
@Nihilistic I have a Matek F722SE running at a 4k loop at only 12% CPU load. Running at 8k is silly with INAV. 8K is for acro mode with Betaflight, not a GPS controlled INAV. The issue is the unrealistic settings set by the user, not the FC.