iNavFlight/inav #3650: I2C MS4525 speed sensor no data

Thyristornp, 2018-07-24:
### Board OMNIBUS F4 Pro, Running firmware released on: Jul 14 2018 22:40:31,  INAV, version: 2.0.0
Speed Sensor on chip MS4525, I2C (Addr. 0x28H)

I connected a speed sensor MS4525 to port J10, but INAV configurator can't detect it (I tried AUTO and MS4525 in setup menu). Then, I conncted I2C bus to logic analizer and there is no any data in it. As I can see, flight controller does not sends requests to speed sensor to take data.

digitalentity, 2018-07-24:
Airspeed is not supported on many boards due to lack of usability.

Thyristornp, 2018-07-25:
So, as I can understand, UAV speed is independent of speed sensor's value?

shellixyz, 2018-07-25:
The speed reported in the OSD is the ground speed (GPS 2D speed)

closed 2019-01-08 by digitalentity