iNavFlight/inav #3651: Wing disarmed by switch instead of RTH

Granger69, 2018-07-24:
### Board and Version
Furious F35 Lightning

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior

Wing went to "disarmed by switch" instead of RTH during signal loss.  I managed to gain control again briefly and manually switched to RTH mode.
 Unfortunately, home point had reset on recovery and so was no good to me.  Switched back to manual and then lost control again. 
 Furious F35 Lightning FC.  Inav 1.9.0.

I might have neglected to Set failsafe on transmitter/receiver after some switch changes.  Any other thoughts on what could have caused this?  Lost the wing in thick brush.  Looking to build another and want to avoid similar occurrence.

Any thoughts?

Skip to 3:35 for issue:

https://www.youtube.com/edit?o=U&video_id=EPSH2gbmwc8

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
CLI Dump:

https://www.dropbox.com/s/gr9d3xp3x2z12ek/Theory%20CLI.txt?dl=0

digitalentity, 2018-07-24:
Known issue. Believed to be fixed in 2.0 - see https://github.com/iNavFlight/inav/pull/3482 and relevant open issues in 2.0 milestone.

Hyprmtr, 2018-07-28:
I had a similar issue. I believe it was because of enabling permanent auto launch and using only Rx for fail safe. I set up FC failsafe(low throt, RTH) and it worked properly again.

Granger69, 2018-07-28:
I really appreciate your input.  I am about to put another wing together and I would like to “know” what happened to the first one.  I did have permanent auto launch enabled.  Although, I had FC failsafe set to RTH.  Also worked when I switched flight mode switch to RTH.  Hmmmm.  Anyway, I will disable Perm Auto Launch on my next wing.  Thanks again.

Hyprmtr, 2018-07-28:
No problem. My issue may be different than yours but, I want to share my experience. I usually always only used Rx fail safe and never had a problem but, it seems I may have to use that in conjunction with FC failsafe in the next FW updates as things are changing. Good luck and good day to you.

digitalentity, 2018-08-05:
Closing. Issue with spontaneous disarm seems to be resolved

closed 2018-08-05 by digitalentity