iNavFlight/inav #3659: new feature request: osd pop up for cruise throttle adjustment.

P-I-Engineer, 2018-07-26:
i added a adjustment for my cruise throttle to be on my throttle stick with a stepped curve in nav modes.

I have cruise throttle enabled on my OSD. I don't really need to look at it all the time. 

would it be possible for the cruise throttle item to pop up on the osd while being adjusted, stay on for a amount of time, then disappear? Could it be shown as a system message?

thanks

shellixyz, 2018-07-27:
The behavior of the RC adjustable values OSD items need stay consistent. A solution would be to use an OSD layout you can switch to when you want to make adjustments containing all the necessary items.

P-I-Engineer, 2018-07-27:
good point on osd layout. I can create what i want using that method.

thanks

closed 2018-10-09 by stale[bot]