iNavFlight/inav #3663: Glide behaviour and speed

DeltaVelocity101, 2018-07-27:
For a recent project, I have been making an aircraft which should attempt to return to home without a motor, how will inav behave under these circumstances? 

With the RTH height set to "at least 10m" and the aircraft being ~15 m off the ground, the aircraft has enough height to traverse and then land. Having said that, how will the speed control algorithm deal with this? Is there any closed loop speed control based on the aircraft attitude or will the flight controller try to throttle up and level out, eventually stalling.

Redshifft, 2018-07-27:
>For a recent project, I have been making an aircraft which should attempt to return to home without a motor, how will inav behave under these circumstances?

badly I would think, you might be able to fudge the settings for it to at least turn back towards home.
I believe there is some work in progress on a descent to home that might get you closer as it's going to fly descending over what we have ATM.

DeltaVelocity101, 2018-07-27:
I'm sorry I don't quite understand, do you mean to say that it would overfly the home point? 

The end goal is to have the aircraft end up above home and loiter its way down to an autoland. So far I've been fudging the settings by not allowing the aircraft to climb (it will be released with plenty of height) and setting the dive angle very low, hopefully forcing it to glide.

Redshifft, 2018-07-27:
You should have explained better what you are doing :) It sounds like you are intentionally wanting a RTH with a glider?
As I said there might be something more suitable with an option to descend on route down the line. the current options are not much good without a motor.

shellixyz, 2018-07-27:
Maybe you can use the new setting `nav_overrides_motor_stop` in 2.0. When RTH is enabled manually with throttle at zero it shouldn't try to maintain altitude. Won't work for RTH triggered by failsafe though.

```
set nav_rth_alt_mode = NO_ALT
set nav_overrides_motor_stop = false
```

digitalentity, 2018-07-28:
@shellixyz I like the idea of `nav_rth_alt_mode` setting.

DeltaVelocity101, 2018-07-28:
@shellixyz in that case will it return to home via a glideslope of sorts?

shellixyz, 2018-07-28:
It will return with the pitch you set into `fw_min_throttle_down_pitch`

closed 2018-10-11 by stale[bot]