iNavFlight/inav #3664: UK legisaltion change - how to meet requirements

ShikOfTheRa, 2018-07-27:
In UK, legislation will be in place in  next few days to make any form of autonomous flight above 400 ft illegal.

for those in organisations, manual flying above is permitted subject to a few other criteria

meh..

So request to add an option to support this....
I guess above this value and to allow normal gyro/stab flying, ONLY if in a mode which is RTL, Waypoints :
MR: start  descent until below 400ft and re-engage mode
FW: stabilised circle. descent until below 400ft and re-engage mode

Any other thoughts? 

Link to changes:
https://bmfa.org/News/News-Page/ArticleID/2533/Update-on-changes-to-UK-Regulations-and-the-400ft-height-limit

giacomo892, 2018-07-27:
Nothing to do within INAV. 
Pilot has the discreption to take off with the machine and respect or not the laws in his country.

stronnag, 2018-07-27:
see also #1510, #2322, #2843

ShikOfTheRa, 2018-07-27:
@giacomo892 
I disagree. It can be to do with iNav - iNav could add support to extend where able to fly legally for UK users.  Current Inav flyers  will be restricted to 400ft only. 

It would unfortunately be end of stock iNav fixedwing for me for sure.
Anyway, let's see if someone else has something more constructive to say.

@Stronnag - yep I get those. Makes sense. I see this as an extension to that

ShikOfTheRa, 2018-07-27:
Unfortunately that suggestion doesn't meet the legislation. It is not permitted to have a device that supports it above 400ft - just changing mode doesn't make it legal. I wish it did....

Doesn't sound like any positive  interest. And I suspect most flyers will probably not care anyway - closing.

Redshifft, 2018-07-27:
let's not get too into all of this. I have always had a problem with the types that need the restrictions and constraints in the first place. These people also are the ones that would fly to the set limits on every occasion regardless of surrounding circumstances etc..
It's a minimum requirement that anyone Using airspace anywhere in the world understands local regulation and always ensures any flight is safe to do so while respecting others.
Saying that Inav requires a lot more intelligence than the muppets flying the ready to fly stuff that has created problems generally possess.

I think it more constructive that we warn and delete videos of people running Inav over heavily populated areas etc, etc, just from the angle of Inav not condoning risky/irresponsible  usage.

@ShikOfTheRa I'm pretty sure most UK users here do care and would have Insurance and common  sense. The way I read it is if your a BMFA member or insured through one of the other organisations you still have dispensation to fly up to 1000feet if safe to do so, the rules regarding spotters are the same and Autonomous flight is pretty much the same grey area it was.

teckel12, 2018-07-27:
@ShikOfTheRa Also, there's no real way for INAV to enforce this as it's a particular altitude above the current position, not the launch position.  Take for example you launch from the base of a mountain, and fly to to top.  Or, launch from the top of a mountain and fly to the base.  INAV would need to know the ground altitude to be able to enforce this, which it doesn't have.  Therefore, this is technically impossible with the hardware that INAV is built on.

digitalentity, 2018-07-28:
Frankly, I'm all in for staying leagal.
What we could do is create a setting to enforce certain regulations and default it to "ON".

digitalentity, 2018-07-28:
I'm reopening it so we could continue the discussion.

teckel12, 2018-07-28:
@digitalentity I see no problem with creating a default 122m altitude limit. Users would be free to set this to anything or 0 to disable.

Probably only a problem for a fixed wing doing a first flight with default setting towards a mountain and not being able to pull up.

Now that I think about it, maybe it shouldn't be set by default.

Redshifft, 2018-07-28:
All we need is terrain following ;)

wx4cb, 2018-07-29:
personally i agree with other posters... the pilot is the one responsible for obeying the laws of the country whatever they're in. 

It's not an INAV problem.

teckel12, 2018-07-29:
And INAV couldn't really enforce an altitude limit because it would need global terrain data which isn't even possible. Better to rely on the pilot to keep legal for the appropriate country and terrain rather than set a limit that will almost always be incorrect (and could cause a crash).

Artyum, 2018-07-31:
Imho even if it would be not a problem to make a variable named: "limit_top_height" it really doesn't matter if it's there or not. Going further, there could be more of these like "limit_max_distance" as you propably not allowed to fly outside LOS in most countries.  Despite it is implemented or not, you should keep your fly legal, so don't use long ranges or use it but not fly away LOS; don't use iNav or use it but keep aircrafl <400ft. It's up to you. It's you who will go to jail, not iNav developers :)

teckel12, 2018-07-31:
@Artyum There also needs to be a `limit_max_distance_perscription` and `limit_max_distance_glasses` to note if I'm wearing my glasses or not, as that greatly changes the LOS distance. Also `limit_max_distance_craft_size` would also be required. Then again...

Artyum, 2018-07-31:
@teckel12 of curse, but according to www.caa.co.uk/Consumers/Unmanned-aircraft/Model-aircraft/ there is also weight limit so don't forget about limit_max_weight..

teckel12, 2018-07-31:
@Artyum However, in all seriousness, creating a geofence for distance and height would be a good idea, as long as it's optional. This could be used for both legal reasons (like altitude and LOS) if the pilot wished, or for beginner pilots.

blurcak, 2018-08-02:
OT: hmm there are speed limit , but didnt see any cars with geofence :) . Its up to BFU to know the rules.

closed 2019-12-24 by DzikuVx