iNavFlight/inav #3666: Eachine TS 215 only has UART3?  Where to connect Mag???

Jawnv71, 2018-07-27:
This eachine TS215 seemed perfect for an iNav conversion with a barometer and a "Customized Omnibus F4 Flight Controller". It flashed just fine,  alt hold works great, but for the life of me I can't figure where to wire in the Compass SCL/SDA and I can’t get any of the NAV modes without the magnetometer hooked up. Any ideas? Can the SCL and SDA share the tx/rx3 with the GPS? (That foam is over the barometer, BTW) Thanks!

[image]

[image]
[image]
[image]

shellixyz, 2018-07-27:
On F4 boards the I²C bus is often shared with UART3 so if only UART3 is broken out you're screwed.

Jawnv71, 2018-07-27:
Bummer. Is there a way to get the nav modes on the configuratior without the magnetometer?  Thanks a lot.

shellixyz, 2018-07-28:
For planes yes but for multi rotors the magnetometer is needed

digitalentity, 2018-07-28:
DJI NAZA GPS is able to communicate both GPS and compass data via a single UART. I think about a small board to sit between GPS/compass and FC and allow using a single UART
CC: @fiam, @DzikuVx

teckel12, 2018-07-28:
@digitalentity Doesn't INAV support a direct DJI NAZA GPS and compass over a single UART without any type of conversion? If he's flashed INAV to the FC, he could connect a DJI NAZA GPS to UART3 without any modifications.

That's at least what the INAV docs suggest.

Jawnv71, 2018-07-28:
Interesting. That sounds like a good idea. Looking at some of the DJI compasses that are available, it appears that there are aftermarket versions with just the four wires. I wonder if there is a clever way to wire that up or if they have some trick piece of hardware in it.

teckel12, 2018-07-28:
@Jawnv71 according to the INAV docs: https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup it shouldn't require anything clever nor any hardware.

Jawnv71, 2018-07-28:
That’s interesting. I still can’t figure out how to wire a 6 wire UBLOX to the 4 wire DJI setup but did find a naza gps for $25 and that’s cheaper than the hassle of swapping FC so I’ll try that.  Thanks again

teckel12, 2018-07-28:
Yeah, just get an actual DJI NAZA GPS is what I was suggesting. I've even seen actual DJI models for about $35.

Jawnv71, 2018-07-29:
Works like a charm.  Thanks again guys!

teckel12, 2018-07-30:
@Jawnv71 Now the problem will be how poorly the GPS works with the interference with the Runcam Swift.

Jawnv71, 2018-07-30:
Oh yea, that came out right away.

closed 2018-10-12 by stale[bot]

Dronefpvpassion, 2020-05-27:
Bonjour je débute das le fpv et suis aussi détenteur d un ts215.
J aimerai intégrer un beitian n-220 sur mon quad mais étant débutant je n ai aucune idée des soudure a faire.
Merci d avance